LibrePCB/LibrePCB #2: Use libxml2 instead of Qt DOM

ubruhin, 2015-07-21:
- save xml files in canonical form (required for version control)
- validate xml files with xsd

use the branch "replace_QtXml_with_libxml2" to implement this

ubruhin, 2016-07-14:
@rnestler Interested to work on this issue? Canonical XML has very high priority because without this, version control of LibrePCB's XML files is a pain...

rnestler, 2016-07-14:
Seems like a isolated enough task so that I won't break to much :wink:

dbrgn, 2016-07-14:
Does libxml protect against XML attacks?

You should probably take a look at https://www.owasp.org/index.php/XML_External_Entity_(XXE)_Prevention_Cheat_Sheet#C.2FC.2B.2B.

rnestler, 2016-07-14:
@dbrgn The link you provided suggests, that it protects against those attacks, doesn't it?

dbrgn, 2016-07-14:
If you set the right options, then probably, yes.

dbrgn, 2016-07-14:
Here's a list of attacks on XML: https://pypi.python.org/pypi/defusedxml#attack-vectors

ubruhin, 2017-04-28:
This should no longer be required because the root problem is solved by #162, so I'll close this issue.

closed 2017-04-28 by ubruhin

rnestler, 2017-04-29:
Is it still desired to use libxml2 in the future?

ubruhin, 2017-04-29:
> Is it still desired to use libxml2 in the future?

Not sure, depends on how good the actual solution works and how libxml2 still can improve LibrePCB.