LibrePCB/LibrePCB #3: Create a library editor

ubruhin, 2015-07-22:
see static library "librepcblibraryeditor"

0xB767B, 2016-09-28:
This is high priority!

ubruhin, 2017-02-26:
This is work-in-progress in #129.

One important question is still open: **How do we prevent the user from breaking existing library elements?** For example removing a pin from an existing symbol would break all components using that symbol.

**My idea is following:**

Each newly created library element is marked as a **draft** (this state is also stored in the elements XML files). All elements marked as drafts cannot be used from other elements (for example a symbol draft cannot be used in a component) and are not visible in schematic/board editors, so they cannot be used in projects.

As soon as the user thinks the library element is (more or less) finished, he needs to release it (i.e. leave the draft state) to make it usable in other library elements and projects. But from this moment on, the library editor disables all features which would break compatibility with the element (e.g. adding/removing pins of a symbol will no longer be possible).

Because the moment of releasing an element is a point-of-no-return, the [undo stack](http://librepcb.org/LibrePCB-Doxygen/master/d2/d8b/classlibrepcb_1_1_undo_stack.html) must be cleared at this time, so undo (CTRL+Z) will not allow to go back to a state before releasing the element.

Maybe this could even be combined with #15: Instead of adding a boolean property `draft`, we could use an "interface hash" as an indicator whether an element is released or not (empty hash = draft, non-empty hash = released).

What do you think @dbrgn?

szg0000, 2017-02-26:
Hi! I think:
>  For example removing a pin from an existing symbol would break all components using that symbol.

I think yes. But if you want to create a new library element: 
1. Open symbol in the library editor.
2. remove a pin from an existing symbol
3. Create a new symbol with e.g. "Save as..."
4. replace the new symbol to the schematic
5. Update PCB.

My idea is to not allow users to overwrite/modify built-in/dedicated symbols.

ubruhin, 2017-02-26:
> But if you want to create a new library element:
> 1. Open symbol in the library editor.
> 2. remove a pin from an existing symbol

Of course this workflow is not possible with a released symbol. But by using a slightly different order it's still no problem:

1. Open symbol in the library editor.
2. Create a new symbol with e.g. "Save as..." --> The new symbol is now a draft
3. Remove a pin from the new symbol
4. Release the new symbol

> My idea is to not allow users to overwrite/modify built-in/dedicated symbols.

And how do they get updates? All library elements (whether official ones or personal ones) needs to be modified from time to time (fix bugs, improvements, add more translations, ...).

szg0000, 2017-02-26:
It's good question. But must have dedicated libraries to make a schematic/layout portable. How do you think to send a schematic in an e-mail to an another workstation?

ubruhin, 2017-02-26:
> But must have dedicated libraries to make a schematic/layout portable.

That's already the case. Every library element which is added to a schematic/board will automatically be copied into the project's directory ([example](https://github.com/LibrePCB/demo-workspace/tree/master/projects/Demo%20Brushless%20Controller/library)) and is from this moment on decoupled from the workspace library.

This makes LibrePCB projects 100% self-contained (i.e. portable), so there are no dependencies to something outside the project's directory.

> How do you think to send a schematic in an e-mail to an another workstation?

Sending a single schematic in source form (i.e. not rendered as PDF/SVG/...) is not possible, you always have to send the whole project. This is done by sending the whole project's directory (for example by using zip/unzip). But you could even use Git for your projects and host them for example on GitHub.

ubruhin, 2017-02-26:
Another option to prevent the user from breaking existing library elements would be much simpler:

Just **allow the user to do everything** (don't disable some features in the library editor), but **show a warning** if the user is about to break a library element (e.g. *This action could break other components, are you really sure to do this?*).

This system may be simpler to understand (no draft state required), but due to the higher responsibility and power of every single user, the likelihood of breaking changes in library elements will probably increase.

This system has also the advantage that there is no point-of-no-return, so the user is also able to fix mistakes even after he thought the element was finished.

szg0000, 2017-02-26:
What about to implement a check box somewhere in the options menu. By default rules are strict, but click out the check box, allow users to do more things with library elements.

dbrgn, 2017-02-26:
How does versioning work currently?

What if "releasing" a footprint increments the version and prevents the user from editing older versions?

ubruhin, 2017-02-26:
> What about to implement a check box somewhere in the options menu.

That should work too. I need to think about it :)

> How does versioning work currently?

Library elements simply have a field where the user can freely choose a version number ;)

> What if "releasing" a footprint increments the version and prevents the user from editing older versions?

That would not help, because the interface of library elements must *never* change, no matter which version number they have.

dbrgn, 2017-02-26:
I see. I think the warning is good enough, as long as it's big and scary :)

PR #129 (2017-09-04): Add library editor
Good news: This adds the long-awaited library editor! :smiley: 
Bad news: It's not yet finished... :disappointed: 

Preview:
[image]

Fixes #3
closed 2017-09-04 by ubruhin