LibrePCB/LibrePCB #6: Design an Icon for LibrePCB

ubruhin, 2015-07-25:
The current icon is downloaded from the internet (it was published under an open source license). It would be nice to have "our own" icon...

ubruhin, 2018-09-04:
@dbrgn proposed to start a logo design contest on https://99designs.com/ and I really like the idea.

Here my suggestion what to choose exactly:

### Category
[App Icon Design](https://99designs.com/app-icon-design) (see [categories](https://99designs.com/categories))

### Package
Silver (or maybe Bronze)
[image]

### Option
Guaranteed, so we have to choose (and pay for) one of the proposed icons.
[image]

### Title
> Icon for LibrePCB, a new OpenSource PCB Design Software

### Description

> We develop an OpenSource electronic PCB design software for hobbyists and small companies creating electronics for personal or commercial use.
>
> We need a logo as an icon for launching the software on Linux, Mac and Windows. It will be used in the start menu, file manager, task bar, title bar, on the website, social media accounts, app stores and so on. Probably we need the icon as SVG, ICO (if possible), ICNS (if possible) and PNG from 16x16 to (at least) 256x256. Maybe the very small icons need to be simplified compared to the higher resolution icons(?).
>
> LibrePCB aims to be a modern and intuitive software, so its logo/icon should also look modern.
>
> We don't have a strong opinion about the content of the icon, but we think it would make sense to see that it's related to printed circuit boards (PCBs), integrated circuits (ICs) or electronics in general.
>
> I attached our current icon. Actually we like it pretty much, but we don't have the copyright on it. So maybe it's worth a try to create a similar icon, but of course we hope to also see some completely new designs :)
>
> For details about our software, please take a look at http://librepcb.org/.

### Any opinions?

dbrgn, 2018-09-05:
Looks good.

Maybe some more explanation on what LibrePCB does (to someone that does not know what PCBs are and how they're developed) would be great, right at the beginning. That helps to get a better picture of the requirements.

I'd drop the requirements for ICO, ICNS and PNG. As long as we get a logo in vector format, we can create those ourselves. Instead, I'd simply state that the logo should look good in 16x16, 48x48, 256x256 and 512x512px (retain the note about simplifications).

dbrgn, 2018-09-05:
Also, as per the [FAQ](https://en.99designs.ch/app-icon-design#faq_header_81585) ("What do I need to prepare for my app icon design contest?") a list of app icons you like would probably help the designers.

Alexander-Wilms, 2018-09-05:
What abou the second option mentioned here: http://support.iconfinder.com/ownership-and-licenses/can-an-icon-be-used-as-a-logo

You could ask the designer of the current icon to create a logo just for LibrePCB.

ubruhin, 2018-09-06:
> Maybe some more explanation on what LibrePCB does (to someone that does not know what PCBs are and how they're developed) would be great, right at the beginning. That helps to get a better picture of the requirements.

I'll try to add it.

> You could ask the designer of the current icon to create a logo just for LibrePCB.

True, but 99designs has the advantage that we (hopefully) get *dozens* of designs, not only one. So I think the chance of getting a very nice icon is much higher than asking only one designer :)

Alexander-Wilms, 2018-09-06:
Right, makes sense.

ubruhin, 2018-09-09:
Survey for some of the already submitted designs: https://99designs.ch/contests/poll/3eb668cd61

Feedback very welcome :smiley:

ubruhin, 2018-09-09:
Survey for the next 8 designs: https://99designs.ch/contests/poll/14ad25c9a6

ubruhin, 2018-09-13:
Final round: https://99designs.ch/contests/poll/339373fea7

ubruhin, 2018-09-15:
Design contest finished, this is our new logo :smiley: 

[image]

closed 2018-09-16 by ubruhin

dbrgn, 2018-09-18:
It's an icon, not a logo :) We could derive a logo from it though.

ubruhin, 2018-09-18:
Sure, I would just add "LibrePCB" next to the icon ;)

rnestler, 2018-09-18:
> It's an icon, not a logo :) We could derive a logo from it though.

Yes that's what I mean :)

> Sure, I would just add "LibrePCB" next to the icon ;)

Yes but the font should be part of the logo.

PR #327 (2018-09-16): Use new logo in all images
Our new logo:

[image]

Fixes #6