LibrePCB/LibrePCB #11: Use XML instead of INI for workspace settings

ubruhin, 2015-07-28:
workspace classes "WorkspaceSettings" and "WSI_Base":

workspace settings are stored in the file "settings.ini" which is created by QSettings. It would be nice to replace the INI file with a XML file, created by "SmartXmlFile" (e.g. "settings.xml").

rnestler, 2016-03-18:
I consider INI a better format for settings than XML. INI is *much* more human readable.

ubruhin, 2016-03-18:
> INI is much more human readable.

Yes, but...
- INI is very limited. With XML it's possible to store data which is more complicated than just a list of key-value pairs.
- In Qt, there is basically no class to work with INI files. It's possible to use QSettings to load/store data from/to INI files, but it's a mess.
- In LibrePCB we have already created some very powerful classes to work with XML files.
- **LibrePCB uses XML for *everything*. I don't like to mix different file formats together.**

rnestler, 2016-07-14:
> In Qt, there is basically no class to work with INI files. It's possible to use QSettings to load/store data from/to INI files, but it's a mess.

We could use something like https://github.com/mayah/tinytoml to read the configuration

dbrgn, 2016-07-14:
I'd also suggest to use a different configuration format than XML. TOML would be nice. XML is a pain to edit by hand (and I think a good configuration format is easily editable by hand).

rnestler, 2016-07-14:
Especially for user preferences which a lot of people store in a "dotfiles" git repository or similar.

ubruhin, 2016-07-14:
> Especially for user preferences which a lot of people store in a "dotfiles" git repository or similar.

This issue is not about the user preferences (e.g. ```/home/user/.config/...```) but workspace preferences which are located in a LibrePCB workspace directory (even in a hidden subdirectory). User settings are handled by QSettings, which is exactly made to deal with user settings on different operating systems.

> XML is a pain to edit by hand

I don't think that anyone needs to edit workspace settings by hand (at least not until now ;). All the workspace settings can easily be changed in the application (any there is no risk to accidentally damage the settings file).

But it's like our other file format discussions: In my opinion it's important to first make it working with existing classes / file formats to not spend too much time for things with low priority. The design of LibrePCB is flexible enough to change the file format at a later time, if this would be required and gets higher priority.

PR #100 (2016-08-05): Refactor workspace settings
Description: See commit messages ;)

Fixes #11
closed 2016-08-05 by ubruhin