LibrePCB/LibrePCB #13: Add clipboard functionality (cut/copy/paste) to schematic/board/library editors

ubruhin, 2015-07-30:


dbrgn, 2018-04-16:
May I nominate this for 0.1? If you have a lot of the same part, you want to multiply that part somehow, including rotation etc.

dbrgn, 2018-04-16:
Actually, never mind, 0.2 is probably fine since you can create boards without this :) It's a mere convenience.

ubruhin, 2018-04-16:
It probably requires really high effort because it's not trivial to implement this. And as it has no effect on the file format, I see no reason to give this high priority (although it would be very useful for users).

alfishe, 2019-05-12:
I would prefer the whole thing documented as user story, but still...

Schematic editor:
Assuming that copy operation affect all selected items (whole schematic page in worst case)
Q1: All selected items copied with their relative positions so layout kept after pasting. Right?
Q2: When paste is activated - where all that item layout should be placed? I.e. how to calculate top-left (or center point) for the pasted content?
Q3: Dealing with different units on source and destination pages / schematics
Q4: Library references to items used during copy / paste
Q5: What if nets on source / destination are different? Or opposite - the same (but we have different GUIDs for nets with same names)

EdizonTN, 2019-10-09:
Please I have another improvement - **snap cursor to grid** in case of use clipboard (as cut/copy origin). It's more powerful. 

My proposal for access: 
1. in case of use the clippborad's by icons (buttons) - use delta as 0
2. In case of use the clipboard's by CTRL+C (shorcut keys) - use delta of cursor aligned to grid. Next in paste with this delta, and use rotate/mirror function, also calculate with this delta (rotate around cursor aligned to grid).

doom-fr, 2020-06-02:
This functionality would be great to tidy up schematics with multiple pages, moving part of schematics from one page to another (now you have to delete and redraw everything.

Regarding that, I do not understand why it is still not developed as it seems to be easy comparing to other functionalities of LibrePCB.
Am I wrong about difficulty ?

ubruhin, 2020-06-02:
FYI, I'm working on this right now :)

> Am I wrong about difficulty ?

Yes, it's really not easy to implement, especially because the content to be copied is not trivial (whole directories for the library elements, a lot of information about the netlist, ...) and when pasting, some special cases need to be handled properly. However, I think it should be available in the next release (although maybe not perfect yet) :slightly_smiling_face:

PR #754 (2020-09-17): Board editor: Implement clipboard cut/copy/paste
Allows to copy&paste board items (even between different projects). In addition, it's even possible to copy graphical objects of a footprint from the library editor into a board.

Limitations:
- Device instances cannot be copied since they would affect the netlist which currently can only be edited by the schematic editor.
- Footprint pads cannot be copied from the library editor into a board since a board can't contain "naked" pads.

Implemented very similar to #724.

ToDo:
- [x] Support cut+paste even for device instances, since this will not  [... cut]
closed 2020-09-17 by ubruhin