LibrePCB/LibrePCB #17: Create a "About LibrePCB" dialog

ubruhin, 2015-08-03:
Replace the current QMessageBox in the ControlPanel with a QDialog which shows some useful information, for example:
- A small description about LibrePCB (inclusive Logo)
- A link to LibrePCB's website (librepcb.org)
- The application's version (maybe with some additional build information like datetime, git branch, ...)
- The license (with a link to http://www.gnu.org/licenses/gpl-3.0.html)
- Maybe a "how to contribute" section (see also #12)

This dialog should be added to the "Help" menu of all main windows (ControlPanel, SchematicEditor, BoardEditor).

dbrgn, 2018-01-23:
Since PR #213 was closed, I plan to give this another attempt with classical UI 🙂

ubruhin, 2018-07-23:
Fixed in #261

closed 2018-07-23 by ubruhin