LibrePCB/LibrePCB #18: Refactoring of the schematic editor state machine

ubruhin, 2015-08-03:
The schematic editor state machine (all classes in the directory [libs/librepcbprojecteditor/schematiceditor/fsm](https://github.com/LibrePCB/LibrePCB/tree/master/libs/librepcbprojecteditor/schematiceditor/fsm)) is only provisional and very buggy (especially the exception handling).

Some refactoring is required to:
- split the FSM into more, but shorter source files
- make the FSM design cleaner and more robust (e.g. exception handling)
- add more features to the schematic editor

The FSM should pass the remo-check before releasing it :smile:

ubruhin, 2016-01-14:
@0xB767B My suggestion for your next steps:
- [ ] Create a new branch (e.g. ```use_boost_statechart_fsm```) and open a new pull request
- [ ] Add [libboost](http://www.boost.org/) to the file [librepcbprojecteditor.pro](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcbprojecteditor/librepcbprojecteditor.pro) in a portable way (this may be pain in the ass because ```libboost``` does not provide ```pkg-config``` support)
- [ ] Start trying to replace the [schematic editor state machine](https://github.com/LibrePCB/LibrePCB/tree/master/libs/librepcbprojecteditor/schematiceditor/fsm) with a [boost statechart fsm](http://www.boost.org/doc/libs/1_60_0/libs/statechart/doc/index.html)

0xB767B, 2016-01-14:
Yes Sir!

0xB767B, 2016-02-12:
We're not going to use boost statechart. Too much overhead. If things get really complicated in the future, we might reconsider using it.

ubruhin, 2016-03-13:
Good News: With #59 the schematic editor FSM is now much simpler! :smiley:

ubruhin, 2019-06-06:
The schematic editor seems to work pretty good, so let's close this.

closed 2019-06-06 by ubruhin