LibrePCB/LibrePCB #21: Library design review/refactoring

ubruhin, 2015-08-21:
The actual library design may be a bit too complicated and may not scale very good.

ubruhin, 2015-08-23:
#### Examples of different library element types:

R/C/L:
- only a few different symbols
- many (!!) different footprints (mostly standard footprints, but also some special footprints)
- for each footprint only a few (or only one) 3D models
- very generic, mostly no exact device (part number) is required

Diodes/Transistors/OpAmps:
- only a few different symbols
- many (!!) different footprints (mostly standard footprints)
- for each footprint only a few (or only one) 3D models
- mostly an exact device (part number) is required (e.g. 1N4007, BC337, LM358N)

LEDs:
- only a few different symbols
- only a few different footprints (mostly standard footprints)
- for each footprint, there should be different 3D models to choose between different colors
- sometimes an exact device (part number) is required, sometimes not

Connectors:
- only a few different symbols
- many (!!) different footprints (some standard footprints, but mostly very special footprints)
- for each footprint only a few 3D models (some connectors are available in different colors)
- often an exact device (part number) is required

Special ICs (FT232, STM32F407, ...):
- mostly a special symbol is required for these devices
- only a few different footprints, sometimes the datasheet describes how they should look like (so specific footprints should be used instead of standard footprints)
- for each footprint only a few (or only one) 3D models

rnestler, 2015-08-24:
Could you also create a pull-request from the branch? Makes discussing the diff easier.

ubruhin, 2015-08-25:
see pull request #22

ubruhin, 2015-08-30:
how to provide / where to save...
- ...suppliers + part numbers required to order components?
- ...prices of components (for each supplier)?
- ...datasheets of components?

ubruhin, 2015-09-05:
how to support different footprints for different soldering processes (e.g. iron, reflow, wave)?
- use separate footprints/packages with different UUIDs?
- support different soldering processes in one footprint/package?

And how to select the preferred soldering process in the board of a project?

ubruhin, 2015-09-07:
One major issue is the file size of library elements:
 * XML files are not optimized for small file size
 * 3D models require lot (!!) of disk space
 * SPICE models require lot of disk space
 * Because of the backward compatibility (copies of older versions), the disk space required by library elements will grow continuously with the time

So, the required disk space of a workspace library will explode with (hundreds of) thousands of library elements.

ubruhin, 2015-11-24:
**At the moment, there are two slightly different proposals for a new library structure:**

|  | Proposal 1 | Proposal 2 |
|-------------------|------------|------------|
| Structure Diagram | branch [library_refactoring](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring/dev/diagrams/svg/library_structure.svg) | branch [library_refactoring_2](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring_2/dev/diagrams/svg/library_structure.svg) |
| Examples Diagram | branch [library_refactoring](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring/dev/diagrams/svg/library_structure_examples.svg) | branch [library_refactoring_2](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring_2/dev/diagrams/svg/library_structure_examples.svg) |
| Description | Devices can have different Packages (see ```Device::PackageVariant```) | Each Device has exactly one Package (no ```Device::PackageVariant```) |

**Advantages of Proposal 1:**
* A device represents (more or less) a whole "device family" - to provide different packages for one component, you only need to create one device (with different packages)
 * Example: A device "LM358" could be available with "DIP-8" or "SOIC-8" package
* As the device represents a whole device family, a SPICE model of the device is available for all package variants of the device (saves disk space)
* In the schematic, it would be possible to specify the device family (e.g. "LM358") without specifying the exact package. The exact package can be different in each board.

**Disadvantages of Proposal 1:**
* More complicated library structure *(too complicated?)*
* Leads to big device XML files as they can contain many different packages
 * Projects will require more disk space as there are many device package variants which are not needed

**Advantages of Proposal 2:**
* The library structure is less complicated!
* Smaller (but more) XML files for devices
 * Example: Instead of one huge "Resistor" device with all possible packages, there would be many small "Resistor" devices (one for each different package)

**Disadvantages of Proposal 2:**
* For each package of a device family (e.g. "LM358"), a separate device must be created, and there is no explicit relation between them (except the common component)
* A SPICE model which is valid for different packages of a device needs to be added to each device separately (so, the same SPICE model exists in different devices --> needs more disk space)

ubruhin, 2015-11-24:
At the moment, our preferred variant is proposal 2, mainly because it is simpler and thus easier to understand. But as a SPICE models cannot be used across different packages/devices (without adding them to the component), this leads to redundancy in the library (and thus requires more disk space).

Maybe it would be better to use SPICE models as separate library elements, which can be used across different components and devices?

By the way, the library design described in proposal 2 can be "easily" upgraded to the design described in proposal 1, if this would be required in the future. The opposite would be much more complicated...

ubruhin, 2015-11-26:
**Added Proposal 3:**

| Proposal 1 | Proposal 2 | Proposal 3 |
|------------|------------|------------|
| [Structure Diagram](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring/dev/diagrams/svg/library_structure.svg) | [Structure Diagram](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring_2/dev/diagrams/svg/library_structure.svg) | [Structure Diagram](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring_3/dev/diagrams/svg/library_structure.svg) |
| [Examples Diagram](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring/dev/diagrams/svg/library_structure_examples.svg) | [Examples Diagram](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring_2/dev/diagrams/svg/library_structure_examples.svg) | [Examples Diagram](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring_3/dev/diagrams/svg/library_structure_examples.svg) |
| Devices can have different Packages (see ```Device::PackageVariant```) | Each Device has exactly one Package (no ```Device::PackageVariant```) | Similar to Proposal 2, but SPICE models are separate library elements |

Proposal 3 is very similar to proposal 2. The only difference is, that SPICE models are now separate library elements instead of embedding them in Components and Devices.

**Advantages (compared with proposal 2):**
 * Reduced redundancy (and thus disk space) as SPICE models can be shared across Components and Devices. Components and Devices need only references to their SPICE models instead of embedding the whole SPICE models.

**Disadvantages (compared with proposal 2):**
 * One more library element --> more complicated? ;)

ubruhin, 2015-12-05:
### New Variants to Compare:
---

#### [Variant 1 (branch library_refactoring_1)](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring_1/dev/diagrams/svg/library_structure.svg):

[image]

#### [Variant 2 (branch library_refactoring_3)](https://github.com/LibrePCB/LibrePCB/blob/feature/library_refactoring_3/dev/diagrams/svg/library_structure.svg):

[image]

ubruhin, 2016-01-03:
fixed with #34 / 2e0dee5f30649531627db031edde615e16e3eacd

closed 2016-01-03 by ubruhin