LibrePCB/LibrePCB #36: Make it possible to close control panel

dbrgn, 2016-01-03:
When using LibrePCB, the control panel stays open even after opening a project. When using a tiling window manager, this is especially impractical :)

[image]

Is it possible to close the control panel without closing all other windows?

ubruhin, 2016-01-03:
> Is it possible to close the control panel without closing all other windows?

No, this is not possible (until now).
The idea behind this concept is that the control panel allows to open as many projects simultaneously as you like. And closing the control panel will close the whole application (with all opened projects)...

Fraserbc, 2022-01-22:
Is there any interest in this as of 2022? I've just started with LibrePCB and it's not great to use it with a tiling window manager. I can work around it by sending the control panel to an unused workspace but it's not optimal

dbrgn, 2022-01-22:
Maybe one way to implement this would be to have a window counter which is incremented for every window. The application is then only terminated when the last window is closed. (I'm not too familiar with Qt window management though, I don't know if this is feasible.)

sparkylein, 2024-04-05:
It looks like the Control Panel is needed to create a new project, open a project or call the Library Manager. 
If the Control Panel would be closed, there is no way to start one of the 3 possibilities mentioned above, even though the schematic editor window or the board editor window is still open.

On macOS LibrePCB could be an application that resides in the statusbar (on left side of the desktop) and could have all the menu items the control panel has (in the menu bar), or just open and close the control panel window...
This can be done very easily in Cocoa (macOS API), but I'm not sure if Qt could, as I never used it before.

Not sure if a tiling window manager like i3 on Linux would be able to handle this though...

closed 2025-05-13 by ubruhin