LibrePCB/LibrePCB #39: Cannot build LibrePCB

dbrgn, 2016-01-03:
I'm getting this error message:

```
[master]danilo@x2000:LibrePCB$ rm -r build
[master]danilo@x2000:LibrePCB$ mkdir build && cd build
[master]danilo@x2000:build$ qmake -r ../librepcb.pro
Reading /mnt/data/Projects/LibrePCB/3rdparty/3rdparty.pro [/mnt/data/Projects/LibrePCB/build/3rdparty]
 Reading /mnt/data/Projects/LibrePCB/3rdparty/gmock/gmock.pro [/mnt/data/Projects/LibrePCB/build/3rdparty/gmock]
Reading /mnt/data/Projects/LibrePCB/libs/libs.pro [/mnt/data/Projects/LibrePCB/build/libs]
 Reading /mnt/data/Projects/LibrePCB/libs/librepcbcommon/librepcbcommon.pro [/mnt/data/Projects/LibrePCB/build/libs/librepcbcommon]
 Reading /mnt/data/Projects/LibrePCB/libs/librepcblibrary/librepcblibrary.pro [/mnt/data/Projects/LibrePCB/build/libs/librepcblibrary]
 Reading /mnt/data/Projects/LibrePCB/libs/librepcbproject/librepcbproject.pro [/mnt/data/Projects/LibrePCB/build/libs/librepcbproject]
 Reading /mnt/data/Projects/LibrePCB/libs/librepcbworkspace/librepcbworkspace.pro [/mnt/data/Projects/LibrePCB/build/libs/librepcbworkspace]
 Reading /mnt/data/Projects/LibrePCB/libs/librepcblibraryeditor/librepcblibraryeditor.pro [/mnt/data/Projects/LibrePCB/build/libs/librepcblibraryeditor]
 Reading /mnt/data/Projects/LibrePCB/libs/librepcbprojecteditor/librepcbprojecteditor.pro [/mnt/data/Projects/LibrePCB/build/libs/librepcbprojecteditor]
Reading /mnt/data/Projects/LibrePCB/librepcb/librepcb.pro [/mnt/data/Projects/LibrePCB/build/librepcb]
RCC: Error in '../../i18n/translations.qrc': Cannot find file 'librepcb_de.qm'
RCC: Error in '../../i18n/translations.qrc': Cannot find file 'librepcb_de_CH.qm'
RCC: Error in '../../i18n/translations.qrc': Cannot find file 'librepcb_gsw_CH.qm'
RCC: Warning: No resources in '../../i18n/translations.qrc'.
Reading /mnt/data/Projects/LibrePCB/tools/tools.pro [/mnt/data/Projects/LibrePCB/build/tools]
 Reading /mnt/data/Projects/LibrePCB/tools/EagleImport/EagleImport.pro [/mnt/data/Projects/LibrePCB/build/tools/EagleImport]
 Reading /mnt/data/Projects/LibrePCB/tools/ProjectLibraryUpdater/ProjectLibraryUpdater.pro [/mnt/data/Projects/LibrePCB/build/tools/ProjectLibraryUpdater]
 Reading /mnt/data/Projects/LibrePCB/tools/UuidGenerator/UuidGenerator.pro [/mnt/data/Projects/LibrePCB/build/tools/UuidGenerator]
 Reading /mnt/data/Projects/LibrePCB/tools/WorkspaceLibraryUpdater/WorkspaceLibraryUpdater.pro [/mnt/data/Projects/LibrePCB/build/tools/WorkspaceLibraryUpdater]
Reading /mnt/data/Projects/LibrePCB/tests/tests.pro [/mnt/data/Projects/LibrePCB/build/tests]
```

There seems to be something wrong with the paths...

Maybe related, I'm getting the following error msg in qtcreator when opening a form:

[image]

The path should actually be /mnt/data/Projects/**LibrePCB/**img/images.qrc.

ubruhin, 2016-01-03:
> RCC: Error in '../../i18n/translations.qrc': Cannot find file 'librepcb_de.qm'

These errors should disappear after building the second time, because the translation files should be created after the build process (maybe one should try to make this working properly...).

Yes, the reason for the error with images.qrc seems to be a wrong path in a *.ui file. I'll check all ui files and fix these wrong paths.

ubruhin, 2016-01-03:
But building should be successful anyway, or don't you get the 'librepcb' executable?

closed 2016-01-03 by ubruhin

dbrgn, 2016-01-03:
Ah, I didn't realize that I can simply continue the process with `make` :)

commit a82265e5 (2016-01-03): controlpanel.ui: fixed wrong filepath to 'images.qrc' (fixes #39)