LibrePCB/LibrePCB #47: improve compnent library handling

0xB767B, 2016-01-22:
Looking at e.g. SES_AddComponents, there's loads of low level magic going on regarding access to the component libray. This must get into to some higher level interface!

ubruhin, 2016-01-22:
I think you mean especially the process of copying components and symbols to the project's library? This is a thing which is even in the low level modules not yet defined very well, so in the higher level modules (e.g. SES_AddComponents) it's a mess as well.

Before improving SES_AddComponents we have to define a clear strategy for copying library elements to the project's library:
* When adding components to a schematic, should we copy all required files from the workspace library to the project library immediately, or only when the user saves the project the next time?
* And on the other side, when removing components from a schematic, should these components be removed from the project's library immediately, or only when the user saves the project the next time?
* When adding/removing immediately: This would lead into changes inside the project's directory even if the user never saves the project --> unacceptable in my opinion!
* When adding/removing only when saving the project: Simple removing the library elements by deleting their directories is not good because this way we may loose some information of that library elements. That's not acceptable too because with the undo command (Ctrl+Z) the removed components must be re-added again (without loosing some information).

I think the only acceptable solution is to save added/removed components to temporary directories, which are synchronized with the project library when saving the project. But this would need lot of work to implement it...

0xB767B, 2016-01-23:
Many thoughts here... But I would suggest to split this discussion into two:
* High level interface used e.g. in SES_AddComponents.
* Handling of workspace/project libraries.

ubruhin, 2016-01-23:
I will fix this issue in the low level modules soon. After that, the SES_AddComponents can be refactored (which is necessary anyway).

ubruhin, 2016-02-15:
I have improved the project library handling in commit 155ba143a8f1383da08a8015ad37a3b54c6bb9f9, but it's still a little bit provisional.

Now there are some new high level undo commands which are important for the schematic/board editor FSM (see directory [libs/librepcbprojecteditor/cmd/](https://github.com/LibrePCB/LibrePCB/tree/master/libs/librepcbprojecteditor/cmd)).

PR #315 (2018-08-29): Refactor ProjectLibrary to fix several issues
- Add support for saving project libraries to backup directory "library~" instead of "library". Now the backup/restore feature of projects should also work properly for the project library. Fixes #47.
- Fix possible error when saving the project library after replacing a library element (remove element and add another one with same UUID). Fixes #312.
- Allow library elements to be saved into empty directories. Sometimes empty directories are not cleaned up properly (for example by Git), so we should handle empty directories just like non-existing d [... cut]
closed 2018-08-29 by ubruhin