LibrePCB/LibrePCB #52: Change directory names within libs. Do not use prefix libreppcb.

0xB767B, 2016-02-12:
It gets more readable.

ubruhin, 2016-02-12:
I agree... But with the prefix, the names of static libraries are more unique. 

For example, includes across static libraries look [like this](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcbproject/library/projectlibrary.cpp) until now:
```c++
#include <librepcbcommon/exceptions.h>
#include <librepcblibrary/sym/symbol.h>
#include <librepcblibrary/cmp/component.h>
```
...because all static libraries have the directory [/libs/](https://github.com/LibrePCB/LibrePCB/tree/master/libs) as include path (```-I``` compiler flag).

Now, when we simple remove the "librepcb" prefix, the includes would look like this:
```c++
#include <common/exceptions.h>
#include <library/sym/symbol.h>
#include <library/cmp/component.h>
```
...which is quite bad because for example [a directory named "library" exists also in the static library "project" ](https://github.com/LibrePCB/LibrePCB/tree/master/libs/librepcbproject/library) (it's the project's library) --> that's confusing and dangerous!

The other way would be to change the include directory of all static libraries to the project's root directory, so the includes would look like this:
```c++
#include <libs/common/exceptions.h>
#include <libs/library/sym/symbol.h>
#include <libs/library/cmp/component.h>
```
...which would fix the problem with ambiguous directory names, but in my opinion that's also not a good solution (why should libs need the project's root directory as include path, even if they only require files from the [/libs/](https://github.com/LibrePCB/LibrePCB/tree/master/libs) directory?).

Any other ideas?

ubruhin, 2016-09-24:
Current directory structure of C++ libraries:
```
LibrePCB/
├── 3rdparty/
│   ├── gmock/
│   ├── hoedown/
│   └── quazip/
└── libs/
    ├── librepcbcommon/
    ├── librepcblibrary/
    ├── librepcblibraryeditor/
    ├── librepcbproject/
    ├── librepcbprojecteditor/
    └── librepcbworkspace/
```

My proposal for a new directory structure:
```
LibrePCB/
└── libs/
    ├── gmock/
    ├── hoedown/
    ├── quazip/
    └── librepcb/
        ├── common/
        ├── library/
        ├── libraryeditor/
        ├── project/
        ├── projecteditor/
        └── workspace/
```

The code example from above would then look like this:
```c++
#include <librepcb/common/exceptions.h>
#include <librepcb/library/sym/symbol.h>
#include <librepcb/library/cmp/component.h>
```

Pros:
 - `3rdparty` is no longer required as an additional include path (`-I` flag).
 - Solves this issue because the prefix "librepcb" becomes a directory and is thus no longer required in the project names.

Cons:
 - 3rd-party libraries are no longer explicitly recognizable as there is no longer a `3rdparty` directory.

rnestler, 2016-09-26:
:+1: Also this way the folder structure is equivalent to the namspaces, isn't it?

ubruhin, 2016-09-26:
> Also this way the folder structure is equivalent to the namspaces, isn't it?

Not exactly, the namespaces of the projects are:
- `librepcb/common` -> `librepcb`
- `librepcb/library` -> `librepcb::library`
- `librepcb/libraryeditor` -> `librepcb::library`
- `librepcb/project` -> `librepcb::project`
- `librepcb/projecteditor` -> `librepcb::project`
- `librepcb/workspace` -> `librepcb::workspace`

rnestler, 2016-09-29:
Would it help to align the folder structure with the namespaces? like:
 - `librepcb/common` → `librepcb::common`
 - `librepcb/library` → `librepcb::library`
 - `librepcb/libraryeditor` → `librepcb::libraryeditor`
 - `librepcb/project` → `librepcb::project`
 - `librepcb/projecteditor` → `librepcb::projecteditor`
 - `librepcb/workspace` → `librepcb::workspace`

ubruhin, 2016-09-30:
> Would it help to align the folder structure with the namespaces? like:

In my opinion, not really. `common` is used in all other namespaces anyway, so we would then add a `using namespace common;` in nearly every file. And the `*editor` projects highly depend on their corresponding logic projects, so we would also add a `using namespace` statement to most source files...

commit 985ca0a3 (2016-10-31): Move/rename all static libraries

- Move all libraries from ./libs/ into ./libs/librepcb/
- Remove prefix "librepcb" from all libraries in ./libs/librepcb/
- Move all libraries from ./3rdparty/ into ./libs/

Fixes #52
PR #127 (2016-11-01): Directory and namespace structure refactoring
There are some issues with the actual directory/namespace structure in this repository:
- Namespaces do not always match the directory structure (for example the static libraries `librepcbproject` and `librepcbprojecteditor` are located in different directories, but both use exactly the same namespace `librepcb::project`).
- The name of the static libraries are quite long (for example `librepcbprojecteditor`), see #52.
- The library manager (which is currently located in `librepcbworkspace`) will depend on `librepcblibraryeditor` in future, but thi [... cut]
closed 2016-11-01 by ubruhin