LibrePCB/LibrePCB #71: ErcMsgDock may be a performance bottleneck

ubruhin, 2016-04-03:
On a decent notebook I saw that the method ```ErcMsgDock::ercMsgAdded()``` is an extreme performance bottleneck. This seems to be caused by calling the method ```QTreeWidgetItem::sortChildren()``` which was very slow.

This is especially bad when performing lot of changes at once, for example when removing all items of a board or schematic. This can lead into a freezing GUI for many seconds.

System where this issue was seen: Intel i7-4712MQ with Ubuntu 14.04 and Qt 5.2.1 from the official Ubuntu Repository.

0xB767B, 2016-11-07:
How about a worker thread?

ubruhin, 2016-11-07:
I think a worker thread is not even required. It should already help to postpone the call to `QTreeWidgetItem::sortChildren()` to a later time, and then executing only once. At the moment, this method is called hundreds or thousands of times when performing large operations in schematics/boards. But updating the ERC list could be done only once after the whole operation.

ubruhin, 2022-03-09:
Never experienced this issue again, so let's close this issue...

closed 2022-03-09 by ubruhin