LibrePCB/LibrePCB #78: Error while preparing SQL query: [...] no such table: components

The-Compiler, 2016-06-12:
Fresh install, new project, when clicking the resistor symbol in the schema editor:

```
[ WARNING ] Could not determine the system's full username! (:0)
[ WARNING ] There was a stale lock on the workspace: "FilePath(/home/florian/LibrePCB-Workspace)" (:0)
[ WARNING ] Could not determine the system's full username! (:0)
[DEBUG-MSG] open project: "/home/florian/LibrePCB-Workspace/projects/test.lpp" (:0)
[ WARNING ] Could not determine the system's full username! (:0)
[ WARNING ] Could not determine the system's full username! (:0)
[DEBUG-MSG] load settings... (:0)
[DEBUG-MSG] settings successfully loaded! (:0)
[DEBUG-MSG] load project library... (:0)
[DEBUG-MSG] successfully loaded 0 symbols (:0)
[DEBUG-MSG] successfully loaded 0 spice models (:0)
[DEBUG-MSG] successfully loaded 0 packages (:0)
[DEBUG-MSG] successfully loaded 0 components (:0)
[DEBUG-MSG] successfully loaded 0 devices (:0)
[DEBUG-MSG] project library successfully loaded! (:0)
[DEBUG-MSG] load circuit... (:0)
[DEBUG-MSG] circuit successfully loaded! (:0)
[DEBUG-MSG] 1 schematics successfully loaded! (:0)
[DEBUG-MSG] 1 boards successfully loaded! (:0)
[DEBUG-MSG] project successfully loaded! (:0)
[EXCEPTION] Error while preparing SQL query: SELECT version, filepath FROM components WHERE uuid = :uuid {SELECT version, filepath FROM components WHERE uuid = :uuid: no such table: components, Unable to execute statement} (library.cpp:708)
terminate called after throwing an instance of 'librepcb::RuntimeError'
  what():  Error while preparing SQL query: SELECT version, filepath FROM components WHERE uuid = :uuid
[1]    13915 abort (core dumped)  ./generated/unix/librepcb
```

remi75, 2016-06-27:
Hello
I had the same problem, 
It seems like it doenst find a database ,

Then after a reboot it kinda worked, without a crash , but the message has changed:
I have a popup window:

"Could not add component:

The component with the UUID "506bd124-6062-400e-9078-b38bd7e1aaee" does not exist in the workspace library!"

It s like there is no library installed,

weirdly , I can load an example project now , but the same message appears when I try 
to add a component .

Regards .

dbrgn, 2016-06-27:
I think so far LibrePCB does not ship with a library yet, so you need to clone and use the demo workspace:

https://github.com/LibrePCB/demo-workspace/

Just clone it and then choose `File > Switch Workspace` in LibrePCB.

The reason why you could open an example project is that - as far as I know - projects copy the used parts into the project directory, to allow users to easily share the projects without depending on a specific library version.

@ubruhin correct me if I'm wrong :) you should probably add a note about the demo workspace to the README.

remi75, 2016-06-27:
yes, you are right .

bad news: when i try to insert a composant , it crashes again .

same error about SQL .

Regards

dbrgn, 2016-06-27:
> @ubruhin correct me if I'm wrong :) you should probably add a note about the demo workspace to the README.

I already added one to the ArchLinux package: https://aur.archlinux.org/cgit/aur.git/tree/librepcb-git.install?h=librepcb-git

ubruhin, 2016-06-28:
> @ubruhin correct me if I'm wrong :) 

Nothing to correct, you're right :)

> you should probably add a note about the demo workspace to the README.

Good point. See #85.

> bad news: when i try to insert a composant , it crashes again . same error about SQL .

Are you sure that the demo workspace was used? This error should not occur if the file `.metadata/v0/library_cache.sqlite` in the workspace directory exists and contains valid data. In the demo workspace, this file is added to the git repository to guarantee that it works properly (in future, this file will be generated automatically, no need to add it to the version control system...).

remi75, 2016-06-28:
now it s ok, i switched WS .

but, the library editor doesnt start.

when i add devices to shematic , with nets :

in the board, we dont see the unrouted wires, hence we cant route them .

is there a rippup fonction ? just see the nets of the other projects ?

regards and thanx for this great projects .

ubruhin, 2016-06-28:
> but, the library editor doesnt start.
> when i add devices to shematic , with nets :
> in the board, we dont see the unrouted wires, hence we cant route them .
> is there a rippup fonction ? just see the nets of the other projects ?

These are all things which are simply not yet implemented, so you need to wait some time ;)
Nevertheless, routing in the board should be possible, even if there are no unrouted wires visible. But the routing feature is very limited until now...

Regards

commit 85f63f3e (2016-07-31): Library: Fix application crash caused by SQL error

When the SQLite database does not exist and the user adds a component to
a schematic, an SQL error appears and the whole application crashes.
The thrown exception is caused by non-existing tables in the database.
The crash is caused by the method Library::getElementFilePathsFromDb()
which is mistakenly marked with the "noexcept" keyword although it can
thow exceptions.

This commit does:
- Avoid the thrown exception by creating all non-existing tables right
  after opening the database (ctor of Library).
- Fix the application crash by correct [... cut]
PR #102 (2016-08-02): Library: Fix application crash caused by SQL error
When the SQLite database does not exist and the user adds a component to
a schematic, an SQL error appears and the whole application crashes.
The thrown exception is caused by non-existing tables in the database.
The crash is caused by the method Library::getElementFilePathsFromDb()
which is mistakenly marked with the "noexcept" keyword although it can
thow exceptions.

This commit does:
- Avoid the thrown exception by creating all non-existing tables right
  after opening the database (ctor of Library).
- Fix the application crash by correcti [... cut]
commit 670c84e4 (2016-08-02): Library: Fix application crash caused by SQL error (#102)

When the SQLite database does not exist and the user adds a component to
a schematic, an SQL error appears and the whole application crashes.
The thrown exception is caused by non-existing tables in the database.
The crash is caused by the method Library::getElementFilePathsFromDb()
which is mistakenly marked with the "noexcept" keyword although it can
throw exceptions.

This commit does:
- Avoid the thrown exception by creating all non-existing tables right
  after opening the database (ctor of Library).
- Fix the application crash by [... cut]
closed 2016-08-02 by ubruhin