LibrePCB/LibrePCB #83: Use imperative commit messages

ubruhin, 2016-06-13:
Even if I like commit messages written in past tense, I think it would be better to use the imperative form, at least because...
- ...most automatically generated merge commit messages (for example from GitHub) use the imperative form
- ...many other projects (including the linux kernel and git itself) use the imperative form
- ...because the imperative form is widely used, many commits in pull requests use the imperative form too (and it's hard to always say "please change your commit message")

Btw, here are nice guidelines for commit messages: http://chris.beams.io/posts/git-commit/

Any thoughts? ;)

dbrgn, 2016-06-13:
:sparkles: :tada: :balloon: :smile:

ubruhin, 2016-06-15:
**From now on, only imperative commit message must be used (no longer past tense)!**

closed 2016-06-15 by ubruhin