LibrePCB/LibrePCB #90: Improve CI build time by using ccache

rnestler, 2016-07-05:
According to https://docs.travis-ci.com/user/caching/#ccache-cache travis supports using ccache, which could speed up builds.

ubruhin, 2016-07-07:
With merging #91, caching is now working for linux builds with gcc. But with clang and/or osx ccache is not working properly, see issues of travis:
- https://github.com/travis-ci/travis-ci/issues/6201
- https://github.com/travis-ci/travis-ci/issues/5383
- https://github.com/travis-ci/travis-ci/issues/5567

I think I will leave this issue open until these issues are solved...

rnestler, 2016-07-07:
What can we do on our side to improve the situation? Are there some workarounds?

ubruhin, 2016-07-07:
I would prefer to just wait some time instead of using an ugly workaround. Maybe this issue gets solved in a few weeks/months, so we don't have to do anything. Leaving this issue open just helps to not forget it ;)

PR #91 (2016-07-07): Use ccache in travis
This could speed up the build. Fixes #90
PR #156 (2017-03-09): Travis: Make ccache also working with clang and OS X
Build times of the first run (caches for clang and osx not yet available):
[image]

Build times after triggering a rebuild (caches from the first run are now available):
[image]

Fixes #90
closed 2017-03-09 by ubruhin