LibrePCB/LibrePCB #92: Use std::unique_ptr<T> instead of raw pointer where applicable

rnestler, 2016-07-07:
There are a lot of places in the code where raw pointers which could be replaced straight forward by `std::unique_ptr`.

ubruhin, 2016-10-06:
Until now, Qt's smart pointers are used instead of the C++11 ones. The only disadvantage seems to be that QScopedPointer has no move semantics. As a workaround we could use QSharedPointer when passing pointers to outside of function scope.

The other way would be to replace all Qt smart pointers with C++11 ones and don't use Qt pointers any more. 

C++11 smart pointers never throw std::exceptions by themself, do they? If they do, we would have to wrap them with our own exception classes...

ubruhin, 2016-11-27:
I consider this as closed for following reasons:
- C++ smart pointers can throw exceptions in many situations. So we would have to wrap all `std::exception` with our own `librepcb::Exception` class. This is PITA.
- Qt's smart pointers are already widely used in LibrePCB. And in every new class I'm creating or refactoring, I use smart pointers instead of raw pointers where applicable.
- Replacing all old raw pointers with smart pointers at once would require lot of effort. I think it's better to replace them step by step when refactoring old classes. Most of the classes will need some refactoring anyway at some point...
- [Using smart pointers is now recommended in the developers documentation](https://github.com/LibrePCB/LibrePCB/commit/97cdea86516fcb4278a511f16737718dbe90ac74#diff-ffaf3dcdbd556e0df700be213ecb89d8R22).
- Maybe Qt provides a unique pointer with move semantics someday ;) Until then we can use `QSharedPointer` instead.

closed 2016-11-27 by ubruhin