LibrePCB/LibrePCB #93: Replace private marked copy/default constructors with =delete

rnestler, 2016-07-07:
In some places in the code the following is found:
```
    private:

        // make some methods inaccessible...
        ErcMsgList();
        ErcMsgList(const ErcMsgList& other);
        ErcMsgList& operator=(const ErcMsgList& rhs);
```
Which should be replaced by
```
public:
        ErcMsgList() = delete;
        ErcMsgList(const ErcMsgList& other) = delete;
        ErcMsgList& operator=(const ErcMsgList& rhs) = delete;
```

ubruhin, 2016-07-07:
Yeah, this issue exists in many header files because in the past I have always marked inaccessible methods ```private```. Later I switched to ```delete```, which I always use now when creating new classes...

ali-alsabbah, 2019-03-15:
I'd like to work on this if mentorship is still available

ubruhin, 2019-03-15:
> I'd like to work on this if mentorship is still available

Sure, go for it! :smiley:  Just ask either @rnestler or me if you have any questions.

rnestler, 2019-06-11:
> I'd like to work on this if mentorship is still available

@ali-alsabbah Are you still interested in this? If yes jut tell us how we can support you achieving it :slightly_smiling_face:

rzvncj, 2021-05-01:
Just browsing here, but wanted to point out that
```c++
public:
        ErcMsgList() = delete;
```
is not really needed, because you already have a user-declared constructor (the copy constructor), so the default constructor will not be generated:

https://en.cppreference.com/w/cpp/language/default_constructor
> _If no user-declared constructors of any kind are provided_ for a class type (struct, class, or union), the compiler will always declare a default constructor as an inline public member of its class.

So what you really need is just:
```c++
public:
        ErcMsgList(const ErcMsgList& other) = delete;
        ErcMsgList& operator=(const ErcMsgList& rhs) = delete;
```

Here's an example:
```c++
class X {

public:
    X(const X&) = delete;
    X& operator=(const X&) = delete;
};

int main()
{
    X x;
}
```
```
$ g++ example.cpp 
constr.cpp: In function ‘int main()’:
constr.cpp:10:7: error: no matching function for call to ‘X::X()’
   10 |     X x;
      |       ^
constr.cpp:4:5: note: candidate: ‘X::X(const X&)’ (deleted)
    4 |     X(const X&) = delete;
      |     ^
constr.cpp:4:5: note:   candidate expects 1 argument, 0 provided
```

PR #887 (2021-07-22): Replace private marked copy operations with = delete
Replace the old style copy-inhibiting idiom of declaring the copy
constructor and copy operator as private (with no implementation)
with the new Scott Meyers recommended one of declaring them public
and deleted.

Fixes #93.
commit 47a96ac1 (2021-07-22): Replace private marked copy operations with = delete

Replace the old style copy-inhibiting idiom of declaring the copy
constructor and copy operator as private (with no implementation)
with the new Scott Meyers recommended one of declaring them public
and deleted.

Fixes #93.
closed 2021-07-22 by ubruhin