LibrePCB/LibrePCB #96: Doxygen Documentation fails when `final` specifier is used

rnestler, 2016-07-11:
Classes with the `final` specifier (For example CategoryTreeItem) have broken Doxygen documentation: http://librepcb.org/LibrePCB-Doxygen/master/d1/d09/classlibrepcb_1_1library_1_1final.html
> final Class Reference

This makes the documentation harder to read and leaves the class search rather broken.

ubruhin, 2016-07-11:
The used doxygen version is too old, this issue was fixed in [version 1.8.2](https://www.stack.nl/~dimitri/doxygen/manual/changelog.html#log_1_8_2).

It seems that our travis config file is wrong. This line:

```
if [ "$QT_BASE" = "55" ]; then sudo apt-add-repository -y ppa:libreoffice/libreoffice-4-2; fi # used to install an up-to-date doxygen
```

does not make sense. I think it should be like following:

```
if [ "$BUILD_DOXYGEN" = "true" ]; then sudo apt-add-repository -y ppa:libreoffice/libreoffice-5-0; fi # used to install an up-to-date doxygen
```

ubruhin, 2016-08-05:
Fixed with #101.

closed 2016-08-05 by ubruhin