LibrePCB/LibrePCB #103: Design an API to retrieve list of available libraries from the internet

ubruhin, 2016-08-05:
Official LibrePCB libraries will be hosted on https://github.com/LibrePCB-Libraries (at least for the beginning). In the LibrePCB application it should be possible to directly browse and download libraries from there without the need to open the web browser (downloading/installing libraries should be as easy as possible).

This could be done by using the [GitHub API](https://developer.github.com/v3/), but their [rate limits for unauthenticated requests](https://developer.github.com/v3/#rate-limiting) may be a problem for us. We could cache the received data locally, but if many LibrePCB application instances are running behind the same public IP address, the problem still exists.

Another solution would be to create our own API running on our own server, e.g. api.librepcb.org. This would give us some advantages:
- No rate limits
- No dependence to GitHub (our Libraries **can** be hosted on GitHub, but **don't have to**).
- The location of our repositories isn't hard-coded in the application. Only the URL of our API server is hardcoded in the application, but the location of the libraries can change in future without breaking compatibility with older LibrePCB versions.
- Possibility of monitoring API requests and building statistics about the usage of LibrePCB ;)
- Higher flexibility in general

For the beginning, a very simple script on the server side should be enough. I think only one command ("give me a list of available libraries") would be required...

Any thoughts?
Someone interested in designing such an API? :)

dbrgn, 2016-08-05:
Using Github could indeed be problematic. See Homebrow, they're using Github and frequently run into speed issues.

I might be interested in creating such an API. It's quite easy to do with http://www.django-rest-framework.org/. We'd have to discuss the specifics though - what resources to we make available through the API, and where do we host them?

It's also very important that alternative API servers can be set up easily. And that our official API is somewhat good regarding availability.

Maybe we could find a sponsor? I'm thinking of something along the lines of https://www.rackspace.com/de-ch, https://www.cloudscale.ch/ or https://www.digitalocean.com/.

dbrgn, 2016-08-05:
There: https://twitter.com/dbrgn/status/761537240149483522

On the bikeshedding side, I still don't like the capital case in neither the organization name nor the repository name :)

ubruhin, 2016-08-05:
Great! Maybe we can discuss that next Monday ;)

dbrgn, 2016-08-07:
> Maybe we can discuss that next Monday ;)

:+1:

dbrgn, 2016-08-07:
OK, we're getting a free VM sponsored by [cloudscale](https://www.cloudscale.ch/)! :sparkles: Thanks!

ubruhin, 2016-08-07:
Great! Let's design the API :D

dbrgn, 2016-08-08:
Example response:

GET /libraries/
```
[
{
    uuid: "9b0d467c-e1b3-42dc-bc20-be8a46b8744a",
    name: {
        "generic": "Microcontrollers",
    },
    url: "git://github.com/librepcb-libraries/,
    last_modified: "2016-08-08T21:50:00+00:00",
    publisher: "LibrePCB",
    dependencies: [
        "10db7994-138c-4a87-b805-35d5716d867c",
        "273081fb-566a-455a-828f-0346624b7552",
    ]
]
```

ubruhin, 2016-08-10:
I have created a new repository for the server-side implementation of the new API: 
https://github.com/LibrePCB/librepcb-api

rnestler, 2016-08-11:
We could do the server in Rust :smiley:

dbrgn, 2016-08-11:
We could alternatively get stuff done :wink:

dbrgn, 2016-08-19:
https://api.librepcb.org/api/v1/

closed 2016-08-19 by dbrgn