LibrePCB/LibrePCB #109: Finally define the workspace/library/repository concept for the first release

ubruhin, 2016-08-18:
## Description
The workspace/library/repository concept is still not completely defined. We need to do that before publishing the first release of LibrePCB. Some discussion (in German) about this topic is available [here](https://etherpad.coredump.ch/p/librepcb), which I will summarize now.

## Workspace Directory Tree
A workspace directory tree will look like this:
```
workspace/
    ├── .librepcb-workspace
    ├── projects/
    │   ├── Project_A/
    │   ├── Project_B/
    │   └── Project_C/
    ├── v0/
    │   ├── metadata/
    │   └── libraries/
    │       ├── local/
    │       │   ├── MyLibrary_A.lplib/
    │       │   ├── MyLibrary_B.lplib/
    │       │   └── MyLibrary_C.lplib/
    │       └── remote/
    │           ├── 5d9abd1b-cf0b-4cf7-8666-20a1add9971e.lplib/
    │           ├── 193ef70d-8dab-4a6c-a672-274c5bf09b68.lplib/
    │           └── c2c427a9-17c6-4400-981c-6ece1c9735c3.lplib/
    ├── v1/
    │   ├── metadata/
    │   └── libraries/
    │       ├── local/
    │       │   ├── MyLibrary_A.lplib/
    │       │   ├── MyLibrary_B.lplib/
    │       │   └── MyLibrary_C.lplib/
    │       └── remote/
    │           ├── 5d9abd1b-cf0b-4cf7-8666-20a1add9971e.lplib/
    │           ├── 193ef70d-8dab-4a6c-a672-274c5bf09b68.lplib/
    │           └── c2c427a9-17c6-4400-981c-6ece1c9735c3.lplib/
    └── v2/
        ├── metadata/
        └── libraries/
            ├── local/
            │   ├── MyLibrary_A.lplib/
            │   ├── MyLibrary_B.lplib/
            │   └── MyLibrary_C.lplib/
            └── remote/
                ├── 5d9abd1b-cf0b-4cf7-8666-20a1add9971e.lplib/
                ├── 193ef70d-8dab-4a6c-a672-274c5bf09b68.lplib/
                └── c2c427a9-17c6-4400-981c-6ece1c9735c3.lplib/
```

### Explanation
- ```.librepcb-workspace```: A (probably empty) file which is only used to determine whether a directory represents a LibrePCB workspace or not.
- ```projects/```: A directory which contains all projects of that workspace. Each project is a subdirectory with their files in it.
- ```v0..*/```: For each file format version, there is a directory which contains the metadata and libraries. A specific application version will only access the metadata/libraries of its corresponding directory, all other ```v*``` directories will be ignored.
  - ```metadata/```: Contains workspace settings and other metadata.
  - ```libraries/local/```: All local libraries (created/maintained by the user itself).
  - ```libraries/remote/```: All libraries which were downloaded from a repository. The library's UUID is used as directory names to avoid name conflicts. These libraries are read-only, the user cannot modify them. To modify them anyway, the user can copy them as a local library and make changes there.

## Upgrade workspace to newer application version
When the user opens the workspace with a specific application's major version the first time (e.g. after updating LibrePCB from ```v1.x``` to ```v2.x```), LibrePCB will realize that the ```v*``` directory (e.g. ```v2```) does not yet exist. Then probably a wizard will be shown to upgrade the workspace to the newer version. Basically, an older version's directory (e.g. ```v1```) will just be copied (e.g. to ```v2```) and the contained files will be upgraded then, if required.

## Downgrade workspace to older application version
As the old ```v*``` directories will not be removed after an upgrade, a downgrade is not required at all. The workspace can thus smoothly be used with different application versions at the same time. If older  ```v*``` directories are no longer required for sure, the user can just remove them (to save disk space).

## Download libraries from a repository
A repository (e.g. [api.librepcb.org](http://api.librepcb.org)) holds a list of available libraries in the internet. The libraries must be hosted as Zip files, for example on the Releases page of a GitHub repository. The application will download these Zip files and extract them into the ```libraries/remote/``` directory in the workspace.

In future we may use Git (or something else) to download/update the libraries, but I think Zip is good enough for the first release of LibrePCB and takes not too much time to implement.

## Contributing features
Contributing to remote libraries will not be available in the first release of LibrePCB because it would take too much time to implement. Advanced users can clone libraries with Git anyway to make commits and pull requests.

## Dependencies of libraries
A library can have dependencies to other libraries. Therefore, a library needs to provide a list of library UUIDs it depends on. We do not track dependencies between each single library element (symbol, component, ...) because that would be too complicated. We also do not take care of the required minimal library version of dependencies, because that would be hard to determine and maintain. The user should just update all his libraries at once to avoid broken dependencies.

This simple system also requires to forbid removing elements from an existing library, because this could break other libraries. So when a library element should no longer be used, it must be marked as "deprecated" instead of removing it.

dns2utf8, 2016-08-18:
Maybe waren the user in Gase LibrePCB detects a Morge recent Version in the workspace.
Warn the user that changes with other LibrePCB versions will not het tracked with this one.

ubruhin, 2016-08-18:
@dns2utf8 Good point :+1:

dbrgn, 2016-08-19:
> In future we may use Git (or something else) to download/update the libraries, but I think Zip is good enough for the first release of LibrePCB and takes not too much time to implement.

Just be careful that upgrading to a different format (e.g. checking out git repos next to ZIPs) won't be difficult in the future.

> Contributing to remote libraries will not be available in the first release of LibrePCB because it would take too much time to implement. Advanced users can clone libraries with Git anyway to make commits and pull requests.

Big +1. Once we have some experience regarding libraries we can still implement contributions using that knowledge.

ubruhin, 2016-08-19:
> Just be careful that upgrading to a different format (e.g. checking out git repos next to ZIPs) won't be difficult in the future.

Yes, that should be quite easy I think. When switching to a different format, the workspace upgrade mechanism can just remove all remove libraries (which were downloaded as Zip files) and check out the same libraries with the new format.

dbrgn, 2016-08-19:
> When switching to a different format, the workspace upgrade mechanism can just remove all remote libraries (which were downloaded as Zip files) and check out the same libraries with the new format.

But that won't be possible if people upload libraries as Zip files to their FTP.

ubruhin, 2016-08-19:
> But that won't be possible if people upload libraries as Zip files to their FTP.

I'm sure we would find a solution for this case, for example:
- The old format will still be usable, only the contributing features remain disabled for such libraries.
- Libraries which are not provided with the new format will no longer be updated unless the new format is used (not the best solution, but may be acceptable).

ubruhin, 2016-08-19:
### New idea
Instead of creating an invisible file called ```.librepcb-workspace``` in the workspace directory, we could even create a visible file like ```LibrePCB-Workspace.lpws``` and associate this file(-extension) with the application LibrePCB. So a file manager will show the LibrePCB icon for this file, and when the user opens that file, LibrePCB will be started with automatically using the desired workspace (the filepath to it will be passed as argument to the application).

dbrgn, 2016-08-19:
Might be useful, yes. But please keep it lowercase :laughing:

ubruhin, 2016-09-02:
### Library directory structure
At the moment, the directory structure of a library looks like this:
```
My_Library.lplib/
├── cmp/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d.cmp/
│       ├── cmp.xml
│       └── .version
├── cmpcat/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d.cmpcat/
│       ├── cmpcat.xml
│       └── .version
├── dev/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d.dev/
│       ├── dev.xml
│       └── .version
├── pkg/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d.pkg/
│       ├── pkg.xml
│       └── .version
├── pkgcat/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d.pkgcat/
│       ├── pkgcat.xml
│       └── .version
├── sym/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d.sym/
│       ├── sym.xml
│       └── .version
├── lplib.xml
└── .version
```

But there are several things I'm thinking about to change:

1. Remove library directory suffix `.lplib`.
  * Advantages:
    * No restrictions on the library directory names, the user can freely choose them
    * Shorter directory names
  * Disadvantages:
    * A directory cannot easily be identified as a LibrePCB library by its directory name. For example on GitHub it would be nice to see the `.lplib` suffix to easily recognize a repository as a LibrePCB library.
2. Remove element directory suffixes (`.cmp`, `.cmpcat`, `.dev`, ...).
  * Advantages:
    * Shorter directory names (only the element UUID, nothing else)
  * Disadvantages:
    * A directory cannot easily be identified as a LibrePCB library element. But as they usually aren't directly visible to the user, this may not be a real disadvantage...
3. Rename `.version` files to a filename which makes it clear what element is contained in the corresponding directory (e.g. `.librepcb-lib`, `.librepcb-cmp`, `.librepcb-cmpcat`, ...).
  * Advantages:
    * If the library element directory suffix is no longer available, the filename of this file shows which type of library element the corresponding directory contains.
    * The content of this file may be extended in future, so that it contains more than only the file version. Then a neutral filename may be better than `.version`.
  * Disadvantages:
    * Longer file names
4. Rename XML files to the same name as their containing XML root node (e.g. `component.xml` instead of `cmp.xml`.
  * Advantages:
    * Don't know, I just like it more because it's the only "human readable" description of a library element ;)
  * Disadvantages:
    * Longer file names

Most of these ideas can be decided separately to either accept or refuse them. If all of them are accepted, the library directory structure would look like this:

```
My_Library/
├── cmp/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── component.xml
│       └── .librepcb-cmp
├── cmpcat/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── component_category.xml
│       └── .librepcb-cmpcat
├── dev/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── device.xml
│       └── .librepcb-dev
├── pkg/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── package.xml
│       └── .librepcb-pkg
├── pkgcat/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── package_category.xml
│       └── .librepcb-pkgcat
├── sym/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── symbol.xml
│       └── .librepcb-sym
├── library.xml
└── .librepcb-lib
```

@rnestler, @dbrgn, @0xB767B  What do you think about this?

ubruhin, 2016-09-02:
Btw, similar questions apply to LibrePCB projects:
- Should a LibrePCB project directory have a suffix like `lpp` to easily recognize them as a LibrePCB project? Like library directories, this would help to easily recognize GitHub repositories as LibrePCB projects (such a repository would then be named like `MyIncrediblePCB.lpp`)
- Should the version file inside a project directory be called `.librepcb-prj` or `.librepcb-project` instead of `.version`?

dbrgn, 2016-09-03:
> 1. Remove library directory suffix `.lplib`.

Totally agree. We can place a certain file inside the directory to make it discoverable.

> 2. Remove element directory suffixes (`.cmp`, `.cmpcat`, `.dev`, ...).

Agree also.

> 3. Rename `.version` files to a filename which makes it clear what element is contained in the corresponding directory (e.g. `.librepcb-lib`, `.librepcb-cmp`, `.librepcb-cmpcat`, ...).

Good idea.

> 4. Rename XML files to the same name as their containing XML root node (e.g. `component.xml` instead of `cmp.xml`.

Yes please :)

Another idea that goes along with 4, is it useful to call directories `cmp`, `cmpcat` and `dev`? That sounds to me like `compare`, `compare category` and `development` :) Why not use the full name (in plural), to make clear what's contained?

```
My_Library/
├── components/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── component.xml
│       └── .librepcb-cmp
├── component_categories/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── component_category.xml
│       └── .librepcb-cmpcat
├── devices/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── device.xml
│       └── .librepcb-dev
├── packages/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── package.xml
│       └── .librepcb-pkg
├── package_categories/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── package_category.xml
│       └── .librepcb-pkgcat
├── symbols/
│   └── 09febfc6-fb2d-4739-ac57-51ac56a01e7d/
│       ├── symbol.xml
│       └── .librepcb-sym
├── library.xml
└── .librepcb-lib
```

ubruhin, 2016-09-03:
> Totally agree. We can place a certain file inside the directory to make it discoverable.

Sure, there would be the file `.librepcb-lib` anyway, but this is only visible when navigating into a repository. Have a look at the [KiCad organization](https://github.com/KiCad). In the list of all repositories you can directly see whether a repository is a footprint library (`.pretty` suffix) or not. In my opinion, this is quite useful ;)

> Another idea that goes along with 4, is it useful to call directories cmp, cmpcat and dev? That sounds to me like compare, compare category and development :) Why not use the full name (in plural), to make clear what's contained?

My idea was just to keep directory names short. The full path to a library element is already quite long, for example:

`/home/user/librepcb-workspace/v0.1/libraries/local/MyLibrary.lplib/cmpcat/09febfc6-fb2d-4739-ac57-51ac56a01e7d/component_category.xml`

and without the abbreviations, they get even longer:

`/home/user/librepcb-workspace/v0.1/libraries/local/MyLibrary.lplib/component_categories/09febfc6-fb2d-4739-ac57-51ac56a01e7d/component_category.xml`

dbrgn, 2016-09-03:
> Have a look at the [KiCad organization](https://github.com/KiCad). In the list of all repositories you can directly see whether a repository is a footprint library (`.pretty` suffix) or not. In my opinion, this is quite useful ;)

I actually hate that :)

> My idea was just to keep directory names short. The full path to a library element is already quite long

That's true, but due to the UUID based directories the file name will be long anyways. On the other hand, I think it would be nice if directories were easily understandable by the user.

@rnestler opinion?

ubruhin, 2016-09-13:
This will be closed soon, see comments in #114.

ubruhin, 2016-09-24:
Closed because #114 is merged.

closed 2016-09-24 by ubruhin