LibrePCB/LibrePCB #110: Pull packages from Conan

dns2utf8, 2016-08-19:
There is a new package manager for C++: https://conan.io/

I hear many **g**ood things lik**e** it is as simple as npm/pip/cargo.
Since it allows bina**r**y distri**b**ution of any kind it could provide a way to include a wrapp**e**d **r**ust library too.

rnestler, 2016-08-23:
There exist a lot of package managers for C++, but none of them has widespread usage...

 - http://www.cpm.rocks/
 - https://github.com/ruslo/hunter
 - http://platformio.org/

But conan really looks promising :smiley:

ubruhin, 2016-09-24:
Such a package manager may also be useful to get `zlib` and `openssl` which are required after merging #122. And `libxml2` will also be required to solve #2.

@dns2utf8 Nice hidden message in your text xD

dns2utf8, 2016-09-25:
:wink: :+1:

ubruhin, 2025-08-03:
The long-term goal is now to move from C++ towards Rust, where the package management problem is solved. Some C++ dependencies were already replaced by Rust alternatives. So let's close this issue.

closed 2025-08-03 by ubruhin