LibrePCB/LibrePCB #111: Run "git describe" only once when compiling LibrePCB

ubruhin, 2016-08-19:
PR #88 has added the output of "git describe" into the about dialog of LibrePCB. But now I just realized that the "git describe" command is executed for every single *.cpp file (~350 times)! That's not really efficient...

ubruhin, 2016-08-22:
@dbrgn Sure that this is easy? How would you solve that? ;)

But after merging #115, the ```git describe``` will only be executed for each file of the subproject ```librepcbcommon``` (instead of the whole project). So this issue automatically becomes less important...

dbrgn, 2016-08-31:
Ah, I didn't realize that this is a part of the compilation step, not of running LibrePCB :)

eduardosm, 2018-11-16:
A possible solution would be to create a new subproject with a single source and a single header, like this:

Header:
```c++
const char *const gitRevision;
```
Source:
```c++
const char *const gitRevision = GIT_COMMIT_SHA;
```

And only define `GIT_COMMIT_SHA` on such subproject (`GIT_COMMIT_SHA` is the old `GIT_VERSION`, isn't it?)

eduardosm, 2018-11-16:
Or maybe using `$$system()` instead of `$(shell )` is a better idea, as described here: https://stackoverflow.com/a/24010395

ubruhin, 2018-11-18:
Or just closing that issue because it's not very important :grin:

ubruhin, 2021-11-05:
Fixed with the migration to CMake -> closing this issue.

closed 2021-11-05 by ubruhin