LibrePCB/LibrePCB #118: Think about release strategy for LibrePCB projects

ubruhin, 2016-09-05:
An EDA software should be able to generate different types of project output, for example:
- Gerber files for PCB manufacturing
- Pick&place data for PCB assembly
- PDF of schematics
- PDF of boards
- Bill of material

My proposal is that each LibrePCB project should have a subdirectory called `generated` or `output` (which one is better?) with a subdirectory for each output type (e.g. `gerber`, `pdf`, ...). LibrePCB will then save all the generated files in their corresponding output directories.

But what should the user then do with the generated files, should he add them to the version control system (e.g. `Git`), or should the VCS ignore the whole output directory ("never check in generated files")?

I think it would be best to also check in all generated files (even if some of them are binary blobs), but only update them just before making a release of the project (e.g. before ordering PCBs). The workflow would then look like this:
 1. Create new LibrePCB project
 2. Work on the project (modify, commit, modify, commit, ...)
 3. Generate output data (gerber, ...)
 4. Commit all the generated data
 5. Create a new tag (e.g. `v1`)
 6. If further modifications are required, continue with step 2)

Any other ideas?

dbrgn, 2016-09-06:
> But what should the user then do with the generated files, should he add them to the version control system (e.g. Git), or should the VCS ignore the whole output directory ("never check in generated files")?

I think that's a matter of opinions. Some people want their output in the repo, others don't.

Can we automatically version the output?

ubruhin, 2016-09-08:
> Can we automatically version the output?

Good question!

We could add a `version` property to project files, which the user can set by himself. Then we can create separate output directories for each version (e.g. `output/v1/gerber/...`, `output/v2/gerber/...`) instead of overwriting the output files with every new release (e.g. `output/gerber/...`).

I see several advantages of such a solution:
- This version number could also be used in schematic frames and board texts (in form of a placeholder). This way, the version of a project is always visible (and automatically updated) in schematics and boards.
- Output files of older releases are always accessible (without checking out git tags). In my opinion it's a common use-case to view older output files (for example schematic PDFs) from time to time. This would be quite painful when output files gets overwritten with each release because this would require to check-out older tags to access these files.

The workflow would then look like this:
 1. Create new LibrePCB project. The version number may be initialized with a default value of `v1`
 2. Work on the project (modify, commit, modify, commit, ...)
 3. Generate output data (gerber, ...). The output goes to `output/v1/...`
 4. Commit all the generated data
 5. Optional: Create a new tag (e.g. `v1`)
 6. If further modifications are required:
    - 6a) Increase the version number of the project
    - 6b) Continue with step 2) -> output goes now to `output/v2/...`

dbrgn, 2016-09-08:
Agree :) Just make sure to keep the version field as a string type. A version could also be "ferienpass 2016".

ubruhin, 2016-09-24:
> Just make sure to keep the version field as a string type. A version could also be "ferienpass 2016".

Done in #114. Because the release strategy is now more clear, I'll close this issue.

closed 2016-09-24 by ubruhin