LibrePCB/LibrePCB #121: Write unit tests for common classes

ubruhin, 2016-09-10:
There are still only a few unit tests. I think first of all, we should write unit tests for the most important common classes, so those located in [libs/librepcbcommon](https://github.com/LibrePCB/LibrePCB/tree/master/libs/librepcbcommon).

Most important unit test (list will be updated regularly):
- [x] DirectoryLock
- [x] FilePath
- [ ] FileUtils _[high priority, easy]_
- [ ] SmartFile, SmartTextFile, SmartVersionFile, SmartXmlFile
- [ ] XmlDomDocument, XmlDomElement _[not yet ready, should be refactored first]_
- [ ] Angle, Length, LengthUnit, Point (already started)  _[high priority, easy]_
- [ ] Alignment _[easy]_
- [x] Application
- [x] ScopeGuard, ScopeGuardList
- [x] SystemInfo
- [ ] UndoCommand, UndoCommandGroup, UndoStack _[high priority]_
- [x] Uuid
- [x] Version

javsanpar, 2019-05-17:
Hello!

I want to learn C++ and I have thought that can practice helping with this issue.

Which tests you think I should start?

Thank you in advance. :smile:

ubruhin, 2019-05-19:
Hi @javsanpar,

> I want to learn C++ and I have thought that can practice helping with this issue.

Great! :smiley: 

> Which tests you think I should start?

I just updated the list in the issue description. You can choose freely which tests to implement, I think `Alignment` or `VersionFile` are easy and thus good units to start with. `FileUtils` requires a bit more effort, and `UndoStack` is probably the most complicated one.

javsanpar, 2019-06-09:
Sorry, I have been doing my final exams and I haven't had time to do the tests until a few days ago. :sweat_smile:

I just did a prototype, I wasn't sure what methods to test.: https://privatebin.net/?c05a7d078ce86f6c#eOwj9zGpDoEP0XjIjKhDA6nb6sbxPmEevmIPbXumWnk=

What do you think?

ubruhin, 2019-06-09:
> Sorry, I have been doing my final exams and I haven't had time to do the tests until a few days ago. sweat_smile

No problem :)

> What do you think?

Looks good so far, except that the test data contains 4 times exactly the same values :grin:

> I wasn't sure what methods to test

The more, the better ;)

Btw, feel free to open a pull request, it's easier to review than an uploaded diff.

Shivam-jainn, 2022-11-20:
Hey ! 
I am learning c++ currently and i have been trying to contribute but never really got around the process ! But since this has a mentor tag , i would love to connect and contribute and also learn at the same time ! 

Thank you

ubruhin, 2022-11-21:
Hi @Shivam-jainn!

I just checked the current state of this topic. I think the only remaining unit where it makes sense to add tests is [`FileUtils`](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcb/core/fileio/fileutils.h).

So it would mean to create a new file `tests/unittests/core/fileio/fileutilstest.cpp` and add at least one test for each of the methods of `FileUtils`. Generally the tests should verify exactly what the class documentation states, for example

- *If the file does not exist, it will be created (with all parent directories).* --> test if the function really creates a new file, even in a directory which doesn't exist.
- *Filepath to a non-existing file (if it exists already, an exception will be thrown).* --> test if there's really an exception thrown when passing a filepath to an existing file.
- ...and so on.

Just have a look at the other unit tests (for example `DirectoryLockTest`) and apply the same coding style to the new test. Also make sure to check out our [contribution guidelines](https://github.com/LibrePCB/LibrePCB/blob/master/CONTRIBUTING.md) and [developers documentation](https://developers.librepcb.org/). Let me know if you have any questions.

BuFran, 2023-09-25:
Hi all! 

I found this great software, thanks for it. It fits exactly to my use case, but still some functionality is missing. I found that this software is open, great! It's documented, very great ! So I will try to help to move forward. 

The good starting point to work, this issue seems to be reasonable, by implementing `FileUtils`/`VersionFile` tests. But the issue text seems to be outdated, units moved (and probably new testable units not described)

At first look, the library is not in `libs/librepcbcommon` (this folder doesn´t exists anyway) but in `libs/librepcb/core`, the description should be updated.

In `fileio` subdir, I found following untested, but testable units:

 * [ ] `FileUtils`, 10 members
 * [ ] `OutputDirectoryWriter`, 8 members (not mentioned)
 * [ ] `VersionFile`, 4 members

So I start work on the simplest `VersionFile` test now (and translation if You accept me as translator).

ubruhin, 2023-09-27:
Hi @BuFran,

Thanks for picking this up! I just updated the first post accordingly.