LibrePCB/LibrePCB #126: Where to store test data files for unit tests?

ubruhin, 2016-10-04:
Some unit tests require test data files to execute their tests (for example zipped libraries, extracted libraries, projects, whatever, ...). But where to store such files?

#### In a subdirectory like "tests/data/" in this repository
Simple, but I don't like to commit many/large/binary files into a repository which contains source code.

#### Create a new repository like "librepcb-test-data" and add it as a submodule to this repository
Better in my opinion ;)

#### Other ideas?

0xB767B, 2016-10-04:
Seperating tests and test data seems not to be a good option to me. How about a new repository for tests and test data?

dbrgn, 2016-10-04:
I agree with @0xB767B that tests and data belong together. On the other hand, the data repo could be added as a submodule, then the reference to a specific version of the test data *is* bundled together with the tests :)

PR #180 (2017-08-21): Tests: Use test data from submodule instead of downloading them from the Internet
Some unit tests downloaded files from the Internet on every run. This was error prone (e.g. network timeouts, SSL certificate errors, ...) and also not really a good design because these tests were (more or less) non-deterministic.

So according to #126 I created the new repository [librepcb-test-data](https://github.com/LibrePCB/librepcb-test-data) where we can store all data required to run the unit tests. That repository is then added as a submodule to `./tests/data` of this repository.

Now the unit tests "do [... cut]
closed 2017-08-21 by ubruhin