LibrePCB/LibrePCB #131: Remove layers for grab areas of symbols and footprints?

ubruhin, 2016-11-06:
Until now we use the schematic layer `SymbolGrabAreas` and the board layers `TopDeviceGrabAreas` and `BottomDeviceGrabAreas` to colorize the grab area of symbols and footprints (in this area you can grab the item with the cursor and move it around). But more and more it turned out that this makes many things much more complicated.

This brings me to the question: **Is it worth it? Or should we just remove that layer?**

### Pros for removing them:
- Less layers visible for the user -> easier to understand
- Less complexity in source code, cleaner design
- Slightly better performance (less semi-transparent areas to paint)

### Pros for keeping them:
- Schematics look better (?)
- Grab areas are visible in schematics and boards
- Optical difference to Eagle schematics (Without grab areas, LibrePCB schematics look nearly identical to Eagle schematics!)

### This is how schematics look with the grab areas:
[image]

### And without grab areas (is this an Eagle schematic?!:wink:):
[image]

ubruhin, 2016-11-06:
### Here an example why grab areas make it difficult to create simple and clear designs:

I want to create a class for the graphical representation of a polygon which can be used in schematics and boards. I name it `PolygonGraphicsItem` and it is derived from [`QGraphicsItem`](http://doc.qt.io/qt-5/qgraphicsitem.html). This polygon class may be used to represent polygons of symbols (in schematics), footprints (in boards), or even something else.

Now, `PolygonGraphicsItem` instances should know the layers which are used to actually paint the polygon. They need to know the color of the outline, the fill color, and whether they are visible or not.

#### If we do not have a separate layer for grab areas, things are very simple:
- `PolygonGraphicsItem` has a method `setLayer(Layer layer)` to set the layer to use.
- `PolygonGraphicsItem` has a method `setIsFilled(bool filled)` to define whether the polygon should be filled or not.
- When painting is requested, `PolygonGraphicsItem` calls `layer.isVisible()`:
  - If it returns `false`, nothing is painted at all.
  - If it returns `true`, the outline is painted with the layer's color. If the polygon needs to be filled, the polygon gets also filled with the layer's color.

#### But if we have a separate layer for grab areas, things get more complicated:
- `PolygonGraphicsItem` has a method `setLineLayer(Layer layer)` to set the layer to use for the outline.
- `PolygonGraphicsItem` has a method `setFillLayer(Layer layer)` to set the layer to use for the area. To fill the area, this is the same as the line layer. To mark the polygon as a grab area, this is the grab-area-layer. To leave the area blank, this is an invalid layer / `nullptr`.
- Outline and area are painted with their defined layers, quite simple... But, stop! What if only one of both layers is visible? Should we really paint the outline and the area separately? Not really, because it would look very strange and does not make sense when grab areas are visible, but symbol outlines not. We could say that the area gets only filled if the outline is also visible, but in my opinion that's an ugly design...

dbrgn, 2016-11-06:
I can't really speak for the code implications, but I kind of like the colored grab areas. Maybe @rnestler can give you more useful feedback :)

ubruhin, 2016-11-06:
Yeah, the more I think about this question, the more I also don't want to remove the grab areas ;) So I think there must be a way to keep this feature.

But another question is, how should these grab area be defined? At the moment, each polygon of a symbol has a flag "is grab area". If this is `true`, the polygon is filled with the yellow layer `SymbolGrabAreas`:
[image]

But now consider a symbol without such a polygon, for example an american resistor:
[image]

Should it be possible to define an explicit grab area for such symbols? But these should then not be painted with the yellow layer because that would look strange. Maybe we should just a property "is visible" to each polygon, which will make the schematic editor hide these polygons?

dbrgn, 2016-11-06:
Maybe we can make the part "glow" or something when hovering over it?

0xB767B, 2016-11-06:
Well, I might be wrong but as I understand, Urban is talking more about the look of the symbols (correct me if I am wrong). So why not get rid of the grab area to make code more simple and have the color of the symbol be a property of the symbol itself (global property or the like). It could be easily made configurable by the user if needed (I now of former colleagues that love light pink ;-) )

rnestler, 2016-11-07:
> Should it be possible to define an explicit grab area for such symbols? But this area should then not be painted with the yellow layer because that would look strange. Maybe we should add a flag "is visible" to each polygon, which will make the schematic editor hide these polygons?

In my opinion the grab area should never be painted *at all*, since as the name just implies, it just defines the area where you can grab the part. If you fill it yellow or whatever should be a property of the outline polygon.

ubruhin, 2016-11-11:
@dbrgn Even if we would use a hover effect, we required a mechanism to define the hovering/grab area ;)

@0xB767B You're right, initially I was talking more about the look of the symbols. But now the question more and more goes also to the functionality of the grab area. For example: how should the grab area be defined with the library editor?

@rnestler Basically I agree with you to completely separate the look of grab areas from their functionality. But consider to create a new symbol in the library editor, this would force you to manually define the grab area with a separate polygon. If *every* polygon would have a property like "is grab area", this is much simpler because you can just use the symbol outline as the grab area too. And in addition, the grab area gets automatically updated if the outline of a symbol gets modified.

ubruhin, 2016-11-12:
### At the moment, this is my favorite system:

**In the symbol editor (which is part of the library editor), the user can draw polygons, either closed or open (polylines). In addition to a list of coordinates, every polygon has following properties:**
- `Width`: The width of the outline.
- `Layer`: The outline of the polygon is filled with this layer's color. If the polygon needs to be filled (see next property), the area is also filled with this layer's color.
- `IsFilled`: If `true`, the area is also filled with the outline layer's color.
- `IsGrabArea`: If true, the whole polygon is also used as the grab area of the symbol.

**Related to the grab areas, there exist two different (instead of only one) grab area layers:**
- `SymbolGrabAreas`: Implicitly used to fill polygons which are defined as grab areas (`IsFilled == false && IsGrabArea == true`). This layer is typically visible in both the symbol editor and schematics.
- `CustomSymbolGrabAreas`: Whole polygons can be associated with this layer to define custom, invisible grab areas. This layer is typically visible in the symbol editor (otherwise the user won't be able to modify its polygons), but not in schematics (the user don't want to see this layer because it would look strange). Actually we could even provide this layer only in the symbol editor, but not in the schematic editor (to avoid confusion by users).

**To manually define an invisible grab area (e.g. for the american resistor), the user can create a polygon with the following properties:**
- `Width`: `0` (indefinitely small)
- `Layer`: `CustomSymbolGrabAreas` _(--> not visible in schematics)_
- `IsFilled`: `true` _(very important, otherwise the polygon gets filled with the layer `SymbolGrabAreas`!)_
- `IsGrabArea`: `true`

rnestler, 2016-11-13:
> Layer: The outline of the polygon is filled with this layer's color. If the polygon needs to be filled (see next property), the area is also filled with this layer's color.

Could you explain the difference / relatiom between polygon and layer? Or link me to the documentation / code? I'm a bit confused...,

ubruhin, 2016-11-13:
> Could you explain the difference / relatiom between polygon and layer? Or link me to the documentation / code? I'm a bit confused...,

|  | Wikipedia | LibrePCB |
|--------------------------|------------------------------------------------------------------|-------------------------------------------------------------------------------------------------------------------------------------------------|
| Polygon (resp. Polyline) | [Polygonal chain](https://en.wikipedia.org/wiki/Polygonal_chain) | [libs/librepcb/common/layer.h](https://github.com/LibrePCB/LibrePCB/blob/add_library_editor/libs/librepcb/common/layer.h) |
| Layer | [Layertechnik](https://de.wikipedia.org/wiki/Layertechnik) | [libs/librepcb/common/geometry/polygon.h](https://github.com/LibrePCB/LibrePCB/blob/add_library_editor/libs/librepcb/common/geometry/polygon.h) |

Does this help? :)

ubruhin, 2017-08-06:
Ok let's go back to this note of @rnestler:

> In my opinion the grab area should never be painted at all, since as the name just implies, it just defines the area where you can grab the part. If you fill it yellow or whatever should be a property of the outline polygon.

Actually I like this strict separation of appearance and functionality. So maybe it would make sense to let the user choose the fill layer of polygons the same way as he chooses the polygon outline layer. To get the yellow areas, he can choose a layer named `SymbolAreas` or similar. But this would *not* automatically define that area as a grab area. There would still be a property `IsGrabArea` which defines the area of a polygon as a grab area.

There would then be a separate layer for grab areas, like `SymbolGrabArea`. This layer is *not* yellow, but for example green or blue to distinguish grab areas from symbol areas. That layer would then only be visible in the symbol editor, but not in the schematic editor, so grab areas will always be completely invisible in schematics.

If the user wants to define the grab area manually (e.g. for the american resistor from above), he can draw a polygon and set both outline and fill layer to `SymbolGrabArea`. Then this grab area is visible in the symbol editor, but not in the schematic editor. But to make it working properly, the property `IsGrabArea` must be set too.

To avoid strange looking elements in schematics and board, the polygons should be filled only if their outline layer is visible too. For example when disabling the layer `SymbolOutlines`, the polygons shouldn't be painted at all, even if the layer `SymbolArea` is enabled (and thus the polygon should be filled).

rnestler, 2017-08-07:
>Wikipedia | LibrePCB
-- | --
Polygon (resp. Polyline) | Polygonal chain | libs/librepcb/common/layer.h
Layer | Layertechnik | libs/librepcb/common/geometry/polygon.h

I just noticed: Did you mix this up in the table or is the Polygon really defined in `layer.h` and the Layer in `polygon.h`? :laughing:

ubruhin, 2017-08-08:
> I just noticed: Did you mix this up in the table or is the Polygon really defined in layer.h and the Layer in polygon.h? :laughing:

That was a test to see if you read my comments xD Fixed it now...

PR #187 (2017-10-08): Add hidden grab area layers
This adds new layers which can be used to draw hidden grab areas (e.g. polygons) in symbols and footprints. These layers are only visible in the library editor (to make them editable), but not in the schematic and board editors.

Details are described in https://github.com/LibrePCB/LibrePCB/issues/131#issuecomment-260089487.

Fixes #131
closed 2017-10-08 by ubruhin