LibrePCB/LibrePCB #137: High frequency layout design needed for LibrePCB board editor

szg0000, 2017-01-04:
High frequency layout design needed for LibrePCB board editor.

szg0000, 2017-01-04:
Would be great for LibrePCB board editor to calculate all the parasitics for the PCB design, and re-generate a new schematic with added parasitic elements, which is good to import to a simulation environment e.g. ngspice.

High frequency design:
1. Draw schematic in LibrePCB schematic editor.
2. Draw layout in LibrePCB board editor. (With 3D modelling, 3D parasitics, 3D spice models)
3. Generate a new schematic with LibrePCB board editor, included the calculated parasitic elements.
4. Import the generated schematic into a simulation environment, e.g. ngspice. Run simulations with the calculated parasitics.

ubruhin, 2017-01-09:
Of course this would be nice, but because there are many other issues with higher priority, it will take years until this feature can be implemented. So I added it to our [wishlist](https://github.com/LibrePCB/LibrePCB/wiki/Wishlist) and close this issue.

closed 2017-01-09 by ubruhin