LibrePCB/LibrePCB #139: Make PPA for Trusty 14.04

Symbian9, 2017-01-18:
Please, make an PPA on Launchpad or on openSUSE Builts services, that would provide DEB's as "nightly builds" for testing.

dbrgn, 2017-01-18:
Yeah, that would actually be pretty cool.

I might give the openSUSE build service a try next Monday, what do you think @ubruhin?

ubruhin, 2017-01-18:
I have already [played around](https://build.opensuse.org/package/show/home:ubruhin/librepcb) with OBS some time ago, but I didn't really understand how it works ;)

> I might give the openSUSE build service a try next Monday, what do you think @ubruhin?

Of course, that would be great!

BTW, I have also played around with [AppImage](http://appimage.org/) on the branch [deploy_appimage](https://github.com/LibrePCB/LibrePCB/tree/deploy_appimage) to upload nightly build packages to [Bintray](https://bintray.com/librepcb)...

dbrgn, 2017-01-18:
If we would be able to use Gitlab CI, we could build all images ourselves using Docker base images... :grin:

probonopd, 2017-01-18:
> BTW, I have also played around with AppImage on the branch deploy_appimage to upload nightly build packages to Bintray...

Works well for me on xubuntu-16.04-desktop-amd64.iso. Thanks @ubruhin

ubruhin, 2017-01-19:
> If we would be able to use Gitlab CI, we could build all images ourselves using Docker base images...

Gitlab CI is awesome, but couldn't we also use Docker on Travis-CI?

> Works well for me on xubuntu-16.04-desktop-amd64.iso.

Great to hear! Unfortunately there are still some issues, I have documented them in #140. Maybe you have some useful tips for us ;)

dbrgn, 2017-01-19:
> Gitlab CI is awesome, but couldn't we also use Docker on Travis-CI?

I don't think so. Travis uses Docker themselves, and for Docker-in-Docker you need a special privileged mode on the host Docker.

probonopd, 2017-01-20:
It is possible to use custom Docker containers on Travis CI. I use it to build some AppImages.

dbrgn, 2017-01-20:
Whoa, really? Do you have a link to the documentation?

probonopd, 2017-01-20:
https://github.com/probonopd/AppImageKit/wiki/Docker-Hub-Travis-CI-Workflow

ubruhin, 2019-06-06:
Since in the meantime we have stable releases available with different deployment methods, and Ubuntu 14.04 is end of life anyway, I think this issue is no longer relevant ;)

closed 2019-06-06 by ubruhin

probonopd, 2019-06-07:
Indeed xenial now is the oldest still-supported LTS release, and there is an AppImage. So, thank you very mcuh :+1: