LibrePCB/LibrePCB #145: Handling of demo workspace when packaging

dbrgn, 2017-02-05:
Right now the demo-workspace is cloned when building the AUR package for Arch Linux (because makepkg clones recursively by default). That's not really necessary and adds to the total download size and time. On the other hand, the user would usually want to set up that workspace.

We can't install it to a system location because then it wouldn't be writeable by the user.

I think the best approach would be to replace the submodule with an initialization script that clones the demo workspace to either the default location (current directory, would be used for CI) or to a location chosen by the user.

The script could then be copied to the target system by the package and instructions to run it could be shown as a post install hook.

What do you think, @ubruhin @rnestler?

ubruhin, 2017-02-05:
I don't like to add an extra build step to clone the demo workspace just to keep the AUR package lightweight. In my opinion, the demo workspace is wanted by everyone which manually clones our git repository, and thus should always be cloned automatically.

I don't see the `librepcb-git` as an "official" way for distributing LibrePCB, so I think it's no problem if it doesn't install very fast (compiling requires a few minutes anyway). To distribute official, stable releases, we should create a new package `librepcb` which only downloads and installs binaries (without a demo workspace).

dbrgn, 2017-02-13:
I agree about your comment, I'll add an installation script for the demo workspace to the package source myself.

closed 2017-02-13 by dbrgn