LibrePCB/LibrePCB #147: Project files (*.lpp) should not contain XML data

ubruhin, 2017-02-11:
LibrePCB project files (`*.lpp`) should be associated with the LibrePCB application to let them open with a simple double click on such a file. On Windows this is usually done by associating the extension `lpp` with LibrePCB. On Unix/Linux there is a more sophisticated mechanism, which can also take the file's content into account to decide which application to open. On OS X I don't know how this is done...

Currently `*.lpp` files are XML files which contain some (meta)data about the project ([example](https://github.com/LibrePCB/demo-workspace/blob/master/projects/Demo%20Brushless%20Controller/Demo%20Brushless%20Controller.lpp)). This is not very smart, because if we switch from XML to another file format some time, this could break the association of the file type with the application. So probably it would be better to remove all project data from this file and store it in another file (e.g. `./core/project.xml`).

But then I wonder what we should write into `*.lpp` files. Maybe only an identifier which can be used for file type detection, like:

```
LIBREPCB-PROJECT
```

Or should we also use the `*.lpp` file to store the file format version, which is currently stored in [`.librepcb-project`](https://github.com/LibrePCB/demo-workspace/blob/master/projects/Demo%20Brushless%20Controller/.librepcb-project)? Then the `.librepcb-project` file could be removed, and the content of the `*.lpp` file is:

```
LIBREPCB-PROJECT
0.1
```

This seems to be quite smart and slightly easier to implement as the file `.librepcb-project` is no longer required. But actually I like the file `.librepcb-project` because of following reasons:

- Its name is always known, so for example it's easy to decide whether a directory contains a LibrePCB project or not. On the other side, the basename of the `*.lpp` file is not known because the user can freely choose it.
- The same mechanism is used for [`.librepcb-lib`](https://github.com/LibrePCB/demo-workspace/blob/master/v0.1/libraries/local/New_Library_Design.lplib/.librepcb-lib), [`.librepcb-sym`](https://github.com/LibrePCB/demo-workspace/blob/master/v0.1/libraries/local/New_Library_Design.lplib/sym/00000000-0000-4001-8000-000000000000/.librepcb-sym) and so on.

To break down the question:
- Should the `*.lpp` file only be a dumb file which makes it possible to open a project with a double click on it (implies that the file is not a mandatory part of a project, so LibrePCB does not care about if and how many of these files exist in the project's directory)?
- Or should the `*.lpp` file be **the** project file which also contains the file format version number (which implies that there must be exactly one `*.lpp` file in a directory, otherwise the project would be invalid)?

@rnestler, @dbrgn, @0xB767B What do you think?

Neonking, 2017-06-15:
Some googling around file formats used in different EDA leaded to this these links :

1. EDIF file format using s expressions : https://en.wikipedia.org/wiki/EDIF
2. IC description : https://en.wikipedia.org/wiki/Caltech_Intermediate_Form
3. IC related db : https://en.wikipedia.org/wiki/GDSII
4. OASIS industry standard : https://en.wikipedia.org/wiki/Open_Artwork_System_Interchange_Standard
5. Data Exchange with ISO 10303 : http://stepcode.org/
6. PADS specification from Mentor Graphics : ftp://ftp.freecalypso.org/pub/CAD/PADS/pdfdocs/Plib_ASCII.pdf
7. ftp://ftp.freecalypso.org/pub/CAD/PADS/pdfdocs/Ppcb_ASCII.pdf
8. KiCAD file spec : file:///C:/Users/Jerome/Downloads/file_formats.pdf

The very intersting Qucs project (http://qucs.sourceforge.net/screenshots.html)  has an interesting discussion around the subject  : https://github.com/Qucs/qucs/issues/475

The best would be to take the best from each to make the best blend! IMO We should developp a custom file format , ASCII readable , based upon maybe Json or S-expr for encapsulating data . I also like very much STEPCODE because of ISO compliance. In general shape the main project file should have an header regarding with project properties, general vars etc... and references to the project subparts .

ubruhin, 2017-09-30:
Currently the project file has following content:

```xml
<project>
 <name>Demo Brushless Controller Project</name>
 <author>U. Bruhin</author>
 <version>v1</version>
 <created>2013-04-13T12:43:52Z</created>
 <schematic>main.xml</schematic> <!-- one entry per schematic file -->
 <board>default.xml</board> <!-- one entry per board file -->
 <attribute key="FOO" type="string" unit="">bar</attribute> <!-- one entry per user-defined attribute -->
</project>
```

I think all these entries, except `schematic` and `board` can be moved into a new separate file `core/metadata.xml`. In the implementation a new class `ProjectMetadata` can be created to hold the project metadata.

But the `schematic` and `board` entries are not really metadata but a list of all existing schematics and boards. Now the question is where this list should be stored. Maybe schematics and boards list should be stored separately, for example into these files:
- `core/schematics.xml` and `core/boards.xml`
- or: `schematics/_index.xml` and `boards/_index.xml`
- or: `schematics/schematics.xml` and `boards/boards.xml`

PR #186 (2017-10-02): Remove XML data in *.lpp project files
- [x] Move project metadata from *.lpp project files into new file `./core/metadata.xml`
- [x] Move schematic and board file lists from *.lpp project file into separate files
- [x] Write mimetype data into *.lpp project files
- [x] Adjust default application assignment for the librepcb mimetype

Fixes #147
closed 2017-10-02 by ubruhin