LibrePCB/LibrePCB #148: ERROR: Could not add component

szg0000, 2017-02-13:
If I want to add a component an error message comes up:

[image]

dbrgn, 2017-02-13:
Did you clone and use the [demo workspace](https://github.com/LibrePCB/demo-workspace)?

szg0000, 2017-02-13:
With demo workspace it works, but I want to use LibrePCB without demo workspace.

dbrgn, 2017-02-13:
I think right now LibrePCB does not ship a basic set of components yet, so those components you tried to add are located inside the demo workspace.

When will we start to split up these components into a standard library, @ubruhin?

rnestler, 2017-02-13:
@szg0000 You could try to download the LibrePCB Base library with the library manager: 
[image]
It should contain the basic elements.

szg0000, 2017-02-13:
Thanks.
I already downloaded that library, but still not works. My library manager version different also.
I "git cloned" it from the master branch. Are you working from the master branch?
[image]

ubruhin, 2017-02-13:
The libraries from the library manager are all empty because the file format is not yet considered as stable. My plan is to move components from the demo workspace into the "official" libraries as soon as the file format is stable. This will be the case after the library editor is in a usable state, which will take at least some months. Until then, the demo workspace is required to use LibrePCB.

szg0000, 2017-02-14:
Is there any discussion or forum topic or meeting, where I can send ideas for the file format development? For schematic and for the board?

ubruhin, 2017-02-15:
No, there is no special forum or similar for such discussions, so it's easiest to just open a new issue here to discuss about the file format. Maybe I'll create a separate repository for RFCs (like [rust-lang/rfcs](https://github.com/rust-lang/rfcs)) some time...

szg0000, 2017-02-15:
A new issue here is enough for comments.

ubruhin, 2017-02-16:
As this problem is clarified, I'll close this issue. Feel free to open a new issue to discuss about the file format.

closed 2017-02-16 by ubruhin

szg0000, 2017-05-10:
I still can't add component. Is the file format clarified?

rnestler, 2017-05-11:
> No, there is no special forum or similar for such discussions, so it's easiest to just open a new issue here to discuss about the file format. Maybe I'll create a separate repository for RFCs (like rust-lang/rfcs) some time...

Actually we have the Telegram group which we sometimes use for discussions: https://t.me/LibrePCB_dev.

ubruhin, 2017-05-21:
> I still can't add component. Is the file format clarified?

No, this is still work in progress. As long as I can't live with the money earned with LibrePCB, I need to go to work the whole week and thus the progress on LibrePCB is quite slow, so you need to be patient :wink: