LibrePCB/LibrePCB #150: LibrePCB file format discussion

szg0000, 2017-02-16:
In this issue we discuss the LibrePCB supported file formats. Feel free to comment.
First, we need a document which describes the latest version of LibrePCB file formats.

szg0000, 2017-02-20:
I see SPICE modelling is under development. Can we invite SPICE developers into LibrePCB project?

I haven't seen any gerber file format description in the documentation. Can we invite linuxcnc.org and http://gerbv.geda-project.org/ developers into LibrePCB project?

dbrgn, 2017-02-20:
Gerber file format should be Gerber X2 (can be found on the website of Ucamco).

I started work on https://github.com/dbrgn/gerber-types-rs. If C++ integration with Rust improves this year (it's on the roadmap) we *might* be able to integrate the library.

szg0000, 2017-02-20:
OK.
What about the UTF-8 support? In the file path, file names, and on the board layer texts, schematic layer texts?

dbrgn, 2017-02-20:
As far as I know this should be covered by QT. @ubruhin probably knows more.

ubruhin, 2017-02-20:
> As far as I know this should be covered by QT.

Right, basically UTF-8 is supported everywhere in the application.

dbrgn, 2017-06-09:
File format discussion: https://github.com/LibrePCB/LibrePCB/wiki/File-Format

ubruhin, 2017-06-11:
S-Expressions look very interesting, so I replaced the XML generator with a quick&dirty S-Expressions generator to compare these two formats:

|           | XML                                                                                                                                                    | S-Expressions                                                                                                                                                 |
|-----------|--------------------------------------------------------------------------------------------------------------------------------------------------------|---------------------------------------------------------------------------------------------------------------------------------------------------------------|
| Symbol    | https://github.com/LibrePCB/demo-workspace/blob/master/v0.1/libraries/local/LibrePCB_Base.lplib/sym/193ef70d-8dab-4a6c-a672-274c5bf09b68/symbol.xml    | https://github.com/LibrePCB/demo-workspace/blob/s-expressions/v0.1/libraries/local/LibrePCB_Base.lplib/sym/193ef70d-8dab-4a6c-a672-274c5bf09b68/symbol.scm    |
| Component | https://github.com/LibrePCB/demo-workspace/blob/master/v0.1/libraries/local/LibrePCB_Base.lplib/cmp/d167e0e3-6a92-4b76-b013-77b9c230e5f1/component.xml | https://github.com/LibrePCB/demo-workspace/blob/s-expressions/v0.1/libraries/local/LibrePCB_Base.lplib/cmp/d167e0e3-6a92-4b76-b013-77b9c230e5f1/component.scm |
| Schematic | https://github.com/LibrePCB/demo-workspace/blob/master/projects/Demo%20Brushless%20Controller/schematics/main.xml                                      | https://github.com/LibrePCB/demo-workspace/blob/s-expressions/projects/Demo%20Brushless%20Controller/schematics/main.scm                                      |
| Circuit   | https://github.com/LibrePCB/demo-workspace/blob/master/projects/Demo%20Brushless%20Controller/core/circuit.xml                                         | https://github.com/LibrePCB/demo-workspace/blob/s-expressions/projects/Demo%20Brushless%20Controller/core/circuit.scm                                         |

For the moment I have chosen the file extension `.scm` (from [Scheme](https://en.wikipedia.org/wiki/Scheme_(programming_language))) to make syntax highlighting working in most editors. [Common Lisp](https://en.wikipedia.org/wiki/Common_Lisp) or other Lisp dialects should also work...

dbrgn, 2017-06-11:
Wow, I really like the general look of the files.

Something I noticed... Usually you would translate this:

```xml
<author>LibrePCB</author>
```

into this:

```scheme
(author "LibrePCB")
```

But in this case:

```xml
<keywords locale="en_US">resistor,resistance</keywords>
```

it becomes

```scheme
(keywords (locale "en_US") (value "resistor,resistance"))
```

The "value" expression is not really necessary, you could convert it to:

```scheme
(keywords (locale "en_US") "resistor,resistance")
```

Then, while parsing, if you get an expression you can treat it as an attribute, while if you get a "plain" value you can treat it as the value.

Another example:

```scheme
(pin (length 2.000000) (rotation 180.000000) (uuid "2b3dd7f8-043b-4d43-9302-9300ba356de7") (x 5.080000) (y 0.000000) (value 2))
```

vs

```scheme
(pin (length 2.000000) (rotation 180.000000) (uuid "2b3dd7f8-043b-4d43-9302-9300ba356de7") (x 5.080000) (y 0.000000) 2)
```

ubruhin, 2017-06-11:
Of course there is room for improvements, I just modified a few lines of code to quick&dirty get some s-expressions files. But if we really decide to switch from xml to s-expressions, there is lot more work to do :wink:

Neonking, 2017-06-15:
Some googling around file formats used in different EDA leaded to this these links :

1. EDIF file format using s expressions : https://en.wikipedia.org/wiki/EDIF
2. IC description : https://en.wikipedia.org/wiki/Caltech_Intermediate_Form
3. IC related db : https://en.wikipedia.org/wiki/GDSII
4. OASIS industry standard : https://en.wikipedia.org/wiki/Open_Artwork_System_Interchange_Standard
5. Data Exchange with ISO 10303 : http://stepcode.org/
6. PADS specification from Mentor Graphics : ftp://ftp.freecalypso.org/pub/CAD/PADS/pdfdocs/Plib_ASCII.pdf / ftp://ftp.freecalypso.org/pub/CAD/PADS/pdfdocs/Ppcb_ASCII.pdf
8. KiCAD file spec : file:///C:/Users/Jerome/Downloads/file_formats.pdf

The very intersting Qucs project (http://qucs.sourceforge.net/screenshots.html)  has an interesting discussion around the subject  : https://github.com/Qucs/qucs/issues/475

The best would be to take the best from each to make the best blend! IMO We should developp a custom file format , ASCII readable , based upon maybe Json or S-expr for encapsulating data . I also like very much STEPCODE because of ISO compliance. In general shape the main project file should have an header regarding with project properties, general vars etc... and references to the project subparts . a json / s expr parser is far more lighter than the xml one too.

Neonking, 2017-06-15:
parsing s expressions : http://people.csail.mit.edu/rivest/sexp.html
IETF draft about s expressions (the most standard ref about it) : http://people.csail.mit.edu/rivest/Sexp.txt

dbrgn, 2017-06-15:
Following that IETF draft might be a good idea. In that case using the reference C implementation might save us some time.

Alternatively if we want to do our own stuff we could also use a library like [sexpresso](https://github.com/BitPuffin/sexpresso/).

Neonking, 2017-06-15:
We also can go for tweaking the original rivest C implementation : http://people.csail.mit.edu/rivest/sexp.html . The advanced form sounds the most usable and readable

ubruhin, 2017-10-10:
So I would say LibrePCB should completely switch from XML to S-Expressions because that file format is *very* simple and seems to provide all the functionality which LibrePCB needs. Or are there any other opinions?

If everyone agrees with switching to S-expressions, I would close this issue and open a new one to track the progress of the file format change and to discuss more details (there *will* be some details to be clarified).

dbrgn, 2017-10-10:
Awesome! I agree :)

itviewer, 2017-10-11:
why not json? 
I think very few people are familiar with S-expressions ,while json is everywhere.
We are developing an EDA tool and decided to use json format，with this https://github.com/nlohmann/json as parser，so far，the conversion of json and Qt data types is very straightforward.

ubruhin, 2017-10-11:
> why not json?

Basically because S-Expressions are more compact and readable. Consider this LibrePCB Symbol definition in S-Expressions:

```scheme
(symbol
 (uuid "00000000-0000-4001-8000-000000000000")
 (version "0.1")
 (author "LibrePCB")
 (name (locale "") "foo")
 (name (locale "gsw_CH") "bar")
 (description (locale "")" foobar")
 (description (locale "gsw_CH") "barfoo")
 (category "00000000-0000-4001-8000-000000000000")
 (pin (uuid "702df32a-635f-46e9-85d2-5b7e2c20aa87") (x -17.78) (y 2.54) (length 5.08) (rotation 0) "1")
 (pin (uuid "1339ffc8-afb2-4137-9d37-d9801ac2bfd0") (x -17.78) (y 0) (length 5.08) (rotation 0) "2")
 (pin (uuid "8207ee9b-89e5-41bc-a633-078952a6c370") (x -17.78) (y -2.54) (length 5.08) (rotation 0) "3")
 (pin (uuid "b5ec7a0f-0f26-482e-a68e-e9a8f108cb80") (x -17.78) (y -5.08) (length 5.08) (rotation 0) "4")
 (polygon (layer "sym_outlines") (width 0.254) (fill false) (grab_area true) (start_x -12.7) (start_y -7.62)
  (segment (angle 0) (end_x -12.7) (end_y 5.08))
  (segment (angle 0) (end_x 12.7) (end_y 5.08))
  (segment (angle 0) (end_x 12.7) (end_y -7.62))
  (segment (angle 0) (end_x -12.7) (end_y -7.62))
 )
 (text (layer "sym_names") (align_v "bottom") (height 2.54) (rotation 0) (x -12.7) (y 5.08) (align_h "left") "#NAME")
 (text (layer "sym_values") (align_v "top") (height 2.54) (rotation 0) (x -12.7) (y -7.62) (align_h "left") "#VALUE")
)
```

compared to JSON (with manually removed linebreaks to keep it shorter):

```json
{
 "uuid": "00000000-0000-4001-8000-000000000000",
 "version": "0.1",
 "author": "LibrePCB",
 "name": {"": "foo", "gsw_CH": "bar"},
 "description": {"": "foobar", "gsw_CH": "barfoo"},
 "category": ["00000000-0000-4001-8000-000000000000"],
 "pin": [
  {"uuid": "702df32a-635f-46e9-85d2-5b7e2c20aa87", "x": "-17.78", "y": "2.54", "length": "5.08", "rotation": "0", "value": "1"},
  {"uuid": "1339ffc8-afb2-4137-9d37-d9801ac2bfd0", "x": "-17.78", "y": "0", "length": "5.08", "rotation": "0", "value": "2"},
  {"uuid": "8207ee9b-89e5-41bc-a633-078952a6c370", "x": "-17.78", "y": "-2.54", "length": "5.08", "rotation": "0", "value": "3"},
  {"uuid": "b5ec7a0f-0f26-482e-a68e-e9a8f108cb80", "x": "-17.78", "y": "-5.08", "length": "5.08", "rotation": "0", "value": "4"}
 ],
 "polygon": [
  {
   "layer": "sym_outlines", "width": "0.254", "fill": "false", "grab_area": "true", "start_x": "-12.7", "start_y": "-7.62", "segment": [
   {"angle": "0", "end_x": "-12.7", "end_y": "5.08"},
   {"angle": "0", "end_x": "12.7", "end_y": "5.08"},
   {"angle": "0", "end_x": "12.7", "end_y": "-7.62"},
   {"angle": "0", "end_x": "-12.7", "end_y": "-7.62"}
  ]
 }],
 "text": [
  {"layer": "sym_names", "align_v": "bottom", "height": "2.54", "rotation": "0", "x": "-12.7", "y": "5.08", "align_h": "left", "value": "#NAME"},
  {"layer": "sym_values", "align_v": "top", "height": "2.54", "rotation": "0", "x": "-12.7", "y": "-7.62", "align_h": "left", "value": "#VALUE"}
 ]
}
```

The S-Expression variant is ~15% shorter (count of characters) and IMO also more readable.

dbrgn, 2017-10-11:
S-Expressions are also *much* simpler to parse. And with a custom generator (which is done in a few lines of code) we can make sure that fields are always written out in the same order, preventing noice in SCM. When switching JSON serializers, the order can change, sometimes even when sticking to the same library.

ubruhin, 2017-10-17:
Further discussion about S-Expressions moved to #192.

closed 2017-10-17 by ubruhin