LibrePCB/LibrePCB #152: Travis-CI: Use docker images

ubruhin, 2017-02-19:
On Travis it's possible to use docker images to build and run the tests, see [Using Docker in Builds](https://docs.travis-ci.com/user/docker/).

Maybe that would be a great solution for LibrePCB as it gives us some advantages:
- Faster CI feedback because the installation of all dependencies (e.g. `apt-get install`) is no longer required (these dependencies come pre-installed in the image). *Needs to be confirmed because I don't know how long Travis requires to pull the image from docker hub.*
- Instead of only installing binaries, it would be even possible to build some tools from source without affecting the CI feedback time. For example [linuxdeployqt](https://github.com/probonopd/linuxdeployqt) may be a candidate for building from source.
- It's easier (resp. even *possible*) to build and run LibrePCB on other distributions than the (very few) ones which are available on Travis. This is extremely helpful for development, testing and deployment!
- Our already existing [Dockerfile](https://github.com/LibrePCB/LibrePCB/blob/master/dev/docker/Dockerfile) would make more sense:
  - It would no longer contain redundant information which is already in the `.travis-ci.yml` file.
  - We would be forced to actively maintain the Dockerfile. At the moment we have no strong motivation to maintain it and thus it's easy to forget adding new dependencies and so on.
  - We would need to upload our image to [Docker Hub](https://hub.docker.com/). This would make using the image much simpler as cloning the LibrePCB repository is no longer required to get the image.

Of course there may appear some technical problems which need to be solved until all of this works properly :wink:

ubruhin, 2017-03-05:
I have just started to play around with Docker Hub. There is now a new repository [LibrePCB/docker-librepcb](https://github.com/LibrePCB/docker-librepcb) which contains the Dockerfiles, and Travis-CI is used to automatically build the images and push them to [Docker Hub](https://hub.docker.com/r/librepcb/librepcb/). Works pretty good! :smiley:

dbrgn, 2017-03-06:
It works, awesome!

rnestler, 2019-06-06:
Is this still relevant with regard to #458 ?

ubruhin, 2019-06-06:
No, it's no longer relevant - CI now uses the Docker images from https://hub.docker.com/r/librepcb/librepcb-dev.

closed 2019-06-06 by ubruhin