LibrePCB/LibrePCB #153: Bundle all fonts used in schematics

ubruhin, 2017-03-06:
### Problem

Depending on the available system fonts, LibrePCB schematics can look different on different operating systems and Linux distributions. This is quite bad, in my opinion the look of schematics must be 100% independent of the used operating system (for boards it would be even more important, but fonts in boards are not yet supported anyway).

### Solution

- [ ] Decide which fonts to use in schematics (need to be released under an open source license).
- [ ] Bundle all required fonts with LibrePCB to make sure they are always available at runtime.
- [ ] Force schematics to always use the bundled fonts instead of system fonts.

szg0000, 2017-03-06:
Is there any open-source font package which is available on all supported platform?

ubruhin, 2017-03-06:
> Is there any open-source font package which is available on all supported platform?

I don't think so. For example on [Alpine Linux](https://www.alpinelinux.org/) it seems that there is not even a single font installed by default (even buttons of LibrePCB don't contain text), and there are only a few fonts available with the package manager.

But even if there is a font available on all supported platforms: If we forget to define a dependency to a font package, or there is another strange problem, most likely the user won't realize that LibrePCB uses wrong fonts. So it's kind of dangerous in my opinion...

dbrgn, 2017-03-06:
Do you need a pixel or a vector font?

szg0000, 2017-03-06:
What about Qt? Is there any Qt supported font packages?

ubruhin, 2017-03-06:
> Do you need a pixel or a vector font?

I would say vector font because scaling is very important in schematics.

> Is there a way to test the look of different fonts in the schematics editor?

Not at runtime. At compile time you could change the font of symbols and netlabels in [SGI_Symbol](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcb/project/schematics/graphicsitems/sgi_symbol.cpp#L52) resp. [SGI_NetLabel](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcb/project/schematics/graphicsitems/sgi_netlabel.cpp#L55).

> What about Qt? Is there any Qt supported font packages?

AFAIK Qt does not provide any fonts, it just lets us use the system fonts.

See also [QFontDatabase](http://doc.qt.io/qt-5/qfontdatabase.html) and [QFont](http://doc.qt.io/qt-5/qfont.html).

dbrgn, 2017-03-06:
> Not at runtime.

We could experiment with different options tonight and create screenshots for comparison :)

ubruhin, 2017-03-08:
[Nimbus Sans L and Nimbus Mono L](https://en.wikipedia.org/wiki/Nimbus_Sans):
[image]

[DejaVu Sans and DejaVu Sans Mono](https://en.wikipedia.org/wiki/DejaVu_fonts):
[image]

[Liberation Sans and Liberation Mono](https://en.wikipedia.org/wiki/Liberation_fonts):
[image]

Beside the look of the fonts, it's also very important to have a font which is well suited for internationalization. I'm not yet sure which font has highest unicode coverage.

See also https://ma.juii.net/blog/free-unicode-font-recommendation

dbrgn, 2017-03-08:
I don't like the monospace version of Nimbus. Can't really decide between Liberation and DejaVu.

What about Fira Sans, Roboto, Noto, Ubuntu or Source Sans Pro? Especially Noto is probably one of the fonts with highest unicode coverage.

ubruhin, 2017-03-08:
I have added some more fonts to my post above.

Noto is quite interesting, but unfortunately it consists of [many separate packages](https://www.google.com/get/noto/). I think we can only support one single font, otherwise the user would need to choose which font to use.

A comparison of the unicode coverage of all these fonts would be interesting, but how can we determine the coverage?

szg0000, 2017-03-08:
You shuold try special characters, like: á,Á,í,Í,Ű,ű,Ő,ő,ü,Ü,ö,Ö,ú,Ú,ó,Ó,é,É
and some chinese and japanese characters.
Can you post screenshots about displaying special charaters, not just the letters of CAN_TX,CAN_RX?

dbrgn, 2017-03-08:
Fira looks quite heavy. Isn't there a lighter font weight available?

dbrgn, 2017-03-08:
Here's a script to test coverage: https://github.com/abelcheung/font-coverage

dbrgn, 2017-03-09:
I don't think we'll get around having to include multiple font files anyways if you want to support CJK in the future :) With around 2000 symbols you already cover latin / greek / cyrillic, that should be enough for most western languages for now.

ubruhin, 2017-03-10:
> Can you post screenshots about displaying special charaters, not just the letters of CAN_TX,CAN_RX?

I will do that as soon as we have a strong idea which fonts we want to use.

> Fira looks quite heavy. Isn't there a lighter font weight available?

It's quite strange, when I zoom into the schematic, the font suddenly switches to a lighter look at a specific zoom level?! No idea why this happens...

> I think so far I prefer Ubuntu and Noto.

One disadvantage of Noto: The monospaced variant "Noto Mono" is only available in regular style, bold and italic variants doesn't exist.

> I don't think we'll get around having to include multiple font files anyways if you want to support CJK in the future :)

OMG, have you seen that the Noto CJK packages are 115MB in size?! I'm not sure if I *ever* want to bundle these packages with LibrePCB ;)

> With around 2000 symbols you already cover latin / greek / cyrillic, that should be enough for most western languages for now.

Of course, but as the fonts are very important, we should still have a plan how to deal with fonts in future. I don't like to switch to another font in (near) future...

szg0000, 2017-03-10:
> Can you post screenshots about displaying special charaters, not just the letters of CAN_TX,CAN_RX?
> 
> I will do that as soon as we have a strong idea which fonts we want to use.

This can help to choose, becuse one font can't display the character correctly, we can avoid that font.

ubruhin, 2018-06-24:
> One disadvantage of Noto: The monospaced variant "Noto Mono" is only available in regular style, bold and italic variants doesn't exist.

I was wrong - the "Noto Mono" was replaced by "Noto Sans Mono", which is also available in bold and italic.

The Noto fonts seem to be really good. They look nice, the Unicode coverage is very good, and the project seems to be very active (see https://github.com/googlei18n/noto-fonts).

So I'd go with Noto Sans and Noto Sans Mono.

Example:

[image]

> You should try special characters, like: á,Á,í,Í,Ű,ű,Ő,ő,ü,Ü,ö,Ö,ú,Ú,ó,Ó,é,É
and some chinese and japanese characters.

[image]

dbrgn, 2018-06-24:
Yes, looks very clean :slightly_smiling_face: :+1:

PR #282 (2018-06-30): Bundle fonts used in schematics together with LibrePCB
This makes LibrePCB using [Noto Fonts](https://www.google.com/get/noto/) for schematics. To ensure consistent look over all platforms, the fonts are bundled together with LibrePCB. So the look of schematics no longer depends on the available system fonts.

Preview:
[image]

Fixes #153.
closed 2018-06-30 by ubruhin