LibrePCB/LibrePCB #157: Decide how to deal with invalid objects

ubruhin, 2017-03-09:
## Problem

### Invalid attributes

There are many possibilities of invalid attributes of objects:
- Empty/invalid strings (for example forbidden characters in library element names)
- Negative numbers for line widths, drill diameters, pad sizes, and many more
- Invalid UUIDs, version numbers, ...
- And many more...

### Invalid combination of attributes

In addition, there are also *combinations* of attributes which I consider as (more or less) invalid:
- Footprints without any visible items (pads, polygons, ...)
- Components without any symbols
- Components which have multiple signals with the same UUID and/or name
- Pin-Signal-Mappings and Pad-Signal-Mappings which contain non-existing pins, pads or signals
- And many more...

Of course some of these invalid states can be avoided by validating new values before applying them, but some things are not really avoidable. For example when creating a new component, it won't have assigned any symbols yet, so there is a component without any symbols. Same for footprints, they are just empty right after creation.

Some things would be possible to avoid, but quite hard to implement. For example a component holds a list of its signals, where all signals must have unique names. But signals do not know each other and thus their method `setName(QString newName)` can't check whether there is already a signal with the same name.

These things are currently validated while saving the elements to their XML file. This was easy to implement, but it's quite annoying for users because they notice the errors only when trying to save the element. And even more annoying, it's currently impossible to save an invalid element, so if you want to stop working on a library element, you either need to fix all errors or discard all changes before you're able to close the library editor.

## Possible Solution

### Invalid attributes

I think it's important to avoid all smaller issues (negative numbers, invalid names, ...) as soon as possible:
1. Whenever possible, initialize attributes with valid, reasonable default values.
2. If 1. is not possible (for example there isn't a reasonable, valid default name for a newly created library element), these attributes must be passed to the constructor of the object to create. The constructor can then validate them and throw an exception if something is invalid.
3. All affected setter methods (for example `setLineWidth(int width)`) must not accept invalid values. They must throw an exception instead, with an expressive message (e.g. "The line width cannot be negative").

### Invalid combination of attributes

As many of these issues cannot be avoided with reasonable effort, maybe it's best to just accept them ;) So saving an invalid object (like a component which contains multiple signals with the same name) is possible. But there should be a way to validate such an element, for example add methods like `bool isValid()` and `QStringList getErrors()`, so the UI is able to display all issues of an invalid element.

@rnestler @dbrgn @0xB767B Any other ideas?

ubruhin, 2017-03-09:
> All affected setter methods (for example setLineWidth(int width)) must not accept invalid values. They must throw an exception instead, with an expressive message (e.g. "The line width cannot be negative").

We could also introduce some new classes which do not allow to represent invalid values. For example in addition to the `Length` class we could create a new class `ULength` which will throw an exception if you try to assign a negative number. So all setter methods which take an `ULength` simply cannot fail, which makes many validity checks and exceptions obsolete.

A class like `ULength` could even be a thin wrapper around a `Length` object which would be used like this:
```
someObject.setLineWidth(ULength(Length::fromString(textField.text())));
```
So the text from the user input would be converted to a `Length` object. That `Length` object is then used to create a `ULength` object from it, while its constructor throws an exception if the passed length was negative. The method `setLineWidth()` could then be declared as `noexcept` because it can never fail.

This way, invalid values are caught in higher-level classes, and lower-level classes don't have to deal with invalid values. One disadvantage is that exception messages are less expressive (e.g. "Value must not be negative" vs. "The line width must not be negative").

dbrgn, 2017-03-09:
I think that's a good idea.

Maybe there's also a good pre-existing implementation for range types / bounded types? http://www.cplusplus.com/forum/general/183337/

> One disadvantage is that exception messages are less expressive

Although you can catch them closer to the source, e.g. when parsing a file. Might be even better in some cases.

rnestler, 2017-03-15:
https://bitbucket.org/davidstone/bounded_integer looks interesting.

closed 2018-08-03 by ubruhin