LibrePCB/LibrePCB #164: Finally define the available graphical layers

ubruhin, 2017-06-08:
Currently the available graphical layers are quite provisional and incomplete. This needs to be fixed before the file format can be marked as stable.

In addition, layers are currently divided into schematic layers and board layers, both with their own Enum containing the layer IDs. Some IDs exist in both schematic and board layers. I think it would make more sense to only work with only one layer type which is used for schematics and boards, with one single Enum containing all these layers (without duplicate IDs).

Btw, the layer ID is a simple unsigned integer and is used to specify the layer of graphical objects. For example a polygon in a schematic symbol has a property `unsigned int layer;`. This ID is then stored in XML files of symbols, footprints, schematics and boards. So they should not change after releasing the first stable version of LibrePCB, otherwise the file format needs to be updated too (which we should avoid).

Here my proposal for all available layers (even if many of them are not yet used):
```
enum ID {

    // 100..199: Schematics
    SchematicBackground         = 100, ///< Primary: background | Secondary: grid
    SchematicSelection          = 105, ///< Primary: outline    | Secondary: area
    SchematicOriginCrosses      = 110, ///< references of symbols, texts, ...
    SchematicSheetFrame         = 115, ///< e.g. A4 sheet frame + text boxes
    SchematicNetLines           = 120, ///< librepcb::project::SI_NetLine
    SchematicNetLabels          = 125, ///< librepcb::project::SI_NetLabel
    SchematicBusLines           = 130, ///< librepcb::project::SI_BusLine
    SchematicBusLabels          = 135, ///< librepcb::project::SI_BusLabel
    // 140..159: Reserved (e.g. for modules / hierarchical sheets)
    SchematicErcAirWires        = 160, ///< unrouted
    SchematicErcWarnings        = 162, ///< markers
    SchematicErcErrors          = 164, ///< markers
    SchematicErcStimulators     = 166, ///< e.g. "pin not connected"-flag
    // 170..179: Reserved
    SchematicDocumentation      = 180, ///< for documentation purposes, e.g. text
    SchematicComments           = 185, ///< for personal comments, e.g. text
    SchematicCustom1            = 190, ///< general purpose layer for the user
    SchematicCustom2            = 191, ///< general purpose layer for the user
    SchematicCustom3            = 192, ///< general purpose layer for the user

    // 200..299: Symbols
    SymbolOutlines              = 200, ///< dark red lines of symbols
    SymbolGrabAreas             = 210, ///< optional yellow area of symbols
    SymbolNames                 = 230, ///< text ${SYM::NAME}
    SymbolValues                = 232, ///< text ${CMP::VALUE}
    SymbolAttributes            = 234, ///< component attribute texts
    SymbolPinLines              = 220, ///< normally same as SymbolOutlines
    SymbolOptionalPinCircles    = 222, ///< green circle of unconnected pins
    SymbolRequiredPinCircles    = 224, ///< red circle of unconnected pins
    SymbolPinNames              = 226, ///< name of the connected component signal
    SymbolPinNumbers            = 228, ///< number of the connected footprint pad

    // 400..499: Footprints/Boards Asymmetrical
    BoardBackground             = 400, ///< Primary: background | Secondary: grid
    BoardSelection              = 405, ///< Primary: outline    | Secondary: area
    BoardOriginCrosses          = 410, ///< references of footprints, holes, ...
    BoardSheetFrame             = 415, ///< e.g. A4 sheet frame + text boxes
    BoardOutlines               = 420, ///< incl. non-plated through hole milling
    BoardAreas                  = 422, ///< area inside of all board outlines
    BoardKeepOut                = 424, ///< keep-out areas for copper and vias
    BoardThtPads                = 430, ///< plated through hole pads
    BoardThtVias                = 435, ///< plated through hole vias
    BoardBlindVias              = 436, ///< plated blind vias
    BoardBuriedVias             = 437, ///< plated buried vias
    BoardNpthDrills             = 440, ///< non-plated through hole drills
    BoardPthMilling             = 445, ///< plated through hole milling
    // 440..459: Reserved (e.g. for modules / hierarchical sheets)
    BoardErcAirWires            = 460, ///< unrouted
    BoardDrcWarnings            = 462, ///< markers
    BoardDrcErrors              = 464, ///< markers
    BoardMeasures               = 470, ///< measurements documentation
    BoardDrillDrawings          = 472, ///< drills documentation (e.g. drill table)
    BoardDrawings1              = 475, ///< special drawings
    BoardDrawings2              = 476, ///< special drawings
    BoardDrawings3              = 477, ///< special drawings
    BoardDocumentation          = 480, ///< for documentation purposes, e.g. text
    BoardComments               = 485, ///< for personal comments, e.g. text
    BoardCustom1                = 490, ///< general purpose layer for the user
    BoardCustom2                = 491, ///< general purpose layer for the user
    BoardCustom3                = 492, ///< general purpose layer for the user

    // 600..699: Footprints/Boards Symmetrical (top/bottom layers can be swapped)
    TopDeviceOutlines           = 600, ///< simple representation of devices
    BottomDeviceOutlines        = 601, ///< simple representation of devices
    TopDeviceGrabAreas          = 604, ///< area where devices can be dragged
    BottomDeviceGrabAreas       = 605, ///< area where devices can be dragged
    TopDeviceOriginCrosses      = 608, ///< references of devices
    BottomDeviceOriginCrosses   = 609, ///< references of devices
    TopDeviceNames              = 612, ///< text, may be used for silk screen
    BottomDeviceNames           = 613, ///< text, may be used for silk screen
    TopDeviceValues             = 616, ///< text, may be used for silk screen
    BottomDeviceValues          = 617, ///< text, may be used for silk screen
    TopOverlay                  = 620, ///< e.g. placement markers on silk screen
    BottomOverlay               = 621, ///< e.g. placement markers on silk screen
    TopSmtPads                  = 624, ///< copper area of smt pads
    BottomSmtPads               = 625, ///< copper area of smt pads
    TopStopMask                 = 630, ///< areas over smt pads
    BottomStopMask              = 631, ///< areas over smt pads
    TopFinish                   = 634, ///< areas of special surface treatments
    BottomFinish                = 635, ///< areas of special surface treatments
    TopSolderPaste              = 640, ///< areas over smt pads
    BottomSolderPaste           = 641, ///< areas over smt pads
    TopGlue                     = 644, ///< adhesive for fixing devices
    BottomGlue                  = 645, ///< adhesive for fixing devices
    TopDeviceKeepOut            = 650, ///< area required to mount devices (courtyard)
    BottomDeviceKeepOut         = 651, ///< area required to mount devices (courtyard)
    TopDevicePlacement          = 660, ///< coordinates for automatic assembly
    BottomDevicePlacement       = 661, ///< coordinates for automatic assembly
    TopTestPoints               = 664, ///< coordinates for automatic pcb testing
    BottomTestPoints            = 665, ///< coordinates for automatic pcb testing

    // 800..899: Board Copper
    TopCopper                   = 800,
    InnerCopper1                = 801,
    InnerCopper2                = 802,
    // [InnerCopper3..96]
    InnerCopper97               = 897,
    InnerCopper98               = 898,
    BottomCopper                = 899,
};
```

szg0000, 2017-06-09:
Maximum how many layers can you define with your proposal?

ubruhin, 2017-06-15:
Mmmh maybe it's not a good idea to identify layers by an integer ID, because it needs to define the IDs very carefully to be ready for future extensions (as you can see above, I left many IDs unused for future extensions).

When using strings as identifiers instead, this limitation does not exist. New layers can be added in future without explicitly reserving an ID (or similar) in the present. And even the files would be much more readable because you don't have to know which ID corresponds to which layer.

So for example this part of a footprint:
```xml
<text layer="612" height="2.54" ...>NAME</text>
<text layer="616" height="2.54" ...>VALUE</text>
<polygon layer="600" width="0.254" ...>
   <segment angle="0" end_x="-12.7" end_y="5.08"/>
   <segment angle="0" end_x="12.7" end_y="5.08"/>
</polygon>
```

would be changed to:
```xml
<text layer="top_device_names" height="2.54" ...>NAME</text>
<text layer="top_device_values" height="2.54" ...>VALUE</text>
<polygon layer="top_device_outlines" width="0.254" ...>
   <segment angle="0" end_x="-12.7" end_y="5.08"/>
   <segment angle="0" end_x="12.7" end_y="5.08"/>
</polygon>
```

Of course in the source code we can keep the enum from above (if required), but then the IDs do not have to be constant across file format versions as they are never stored in files.

> Maximum how many layers can you define with your proposal?

Up to 100 copper layers. But by using strings instead of integers, it should be easy to extend this number in future. So we could for example start with 32 copper layers and add more layers as soon as required.

dbrgn, 2017-06-15:
Yeah, that's probably not a bad idea. The tradeoff regarding storage size is fine I guess.

PR #174 (2017-08-23): Refactor schematic and board layers
- Replace classes `SchematicLayer` and `BoardLayer` by the new class `GraphicsLayer`
- Layers are now identified by strings instead of numbers
- Board: Allow to specify the count of inner copper layers
- Board: Save user settings in separate file (e.g. layer colors are no longer tracked by VCS)

Fixes #164.
closed 2017-08-23 by ubruhin