LibrePCB/LibrePCB #166: Provide an OpenBSD port of LibrePCB

ubruhin, 2017-06-09:
https://www.openbsd.org/faq/faq15.html#Ports

Neonking, 2017-06-15:
https://www.openbsd.org/faq/ports/guide.html
http://man.openbsd.org/bsd.port.mk
http://man.openbsd.org/ports.7
http://man.openbsd.org/make.1

A look for the string "qmake" on http://man.openbsd.org/port-modules reveals qmake options for compilation / porting , no need for any other makefile nor cmake build system.

ubruhin, 2021-12-07:
LibrePCB is now already spread enough that it is even picked up automatically by independent package maintainers. So I think this is no longer a task which needs to be done by LibrePCB developers. It makes more sense to contact package maintainers directly, so let's close this issue here.

closed 2021-12-07 by ubruhin

dante-robinson, 2023-07-12:
Would love to see this working even just compiling manually. I get the error "Could not find a package configuration file provided by "Qt5"" when trying to compile with cmake with qt5 installed.

ubruhin, 2023-07-12:
Hi @dante-robinson, generally LibrePCB should compile fine on OpenBSD but I don't know exactly how to invoke the build commands. [Here](https://openports.pl/path/x11/qt5/qtbase) I found a hint that the Qt installation on OpenBSD appears to be different than on other systems, so you might need to tell cmake where Qt is installed. Unfortunately I can't help with that since I'm not familiar with OpenBSD.

You could try to ask in our [chat or forum](https://librepcb.org/help/) if someone knows more. [Here](https://librepcb.discourse.group/t/building-on-openbsd/320) someone was able to build LibrePCB, but it's outdated because in the mean time we switched from qmake to cmake.

If you get it working, please let me know so we can add the instructions to our documentation!