LibrePCB/LibrePCB #168: Wishlist

0xB767B, 2017-06-23:
Lets have this issue to submit entries for the wishlist so the master of disaster can add them to the wishlist in the wiki.

0xB767B, 2017-06-23:
Plaese add this entry to the wishlist.

Schematic edit should have the possibility to draw high level schematics. The blocks/components should provide a way to enter SystemC-code. Furthermore an export for the complete SystemC-code should be available.

ubruhin, 2017-06-23:
Done -> https://github.com/LibrePCB/LibrePCB/wiki/Wishlist

EdizonTN, 2019-07-18:
**Import/Export DRC settings (Eagle/Altium/KiCAD or other format also)**

- will be good feature. Most PCB FAB houses, provide their ability in this format. Is easy, import this settings into LibrePCB. Example: https://www.eurocircuits.com/drc-settings-and-guide-lines-for-cad-packages/

tinkercnc, 2021-04-08:
### Import of Gerber Files
It would shrinking the time to reanimate old projects made with other pcb tools. :)

ubruhin, 2023-03-30:
Closing this issue because the wishlist is not actively used. Let's just create a separate issue for each idea to avoid having different ways of feature requests.