LibrePCB/LibrePCB #171: Review/Refactor Attributes System

ubruhin, 2017-07-27:
LibrePCB has a quite sophisticated attribute system which allows to display texts in schematics and boards which are dynamically determined. For example in a symbol one can place a label with the text "${SYM::NAME}". As soon as such a symbol is added to a schematic, the text is replaced by the symbol's name (e.g. "R5").

But actually I'm not sure if this concept is really good:
- Maybe the syntax "${NAMESPACE::ATTRIBUTE}" could be simplified (IMO it looks ugly)?
- Do we really need to separate namespace (e.g. "SYM") and attribute (e.g. "NAME")? Maybe we could completely remove the namespace. If still required, the namespace could be integrated into the attribute name instead (e.g. "${SYM_NAME}").
- Should we introduce a strict separation between built-in attributes and user defined attributes? For example the attribute "${SYM::NAME}" is a built-in attribute which is implicitly defined by symbols, it's not listed in any attribute list which the user can edit. If the user would add an attribute "NAME" to a symbol, we actually would have a name conflict as there is already a built-in attribute with the same name! But how could a strict separation look like?

dbrgn, 2017-07-31:
> Maybe the syntax "${NAMESPACE::ATTRIBUTE}" could be simplified (IMO it looks ugly)?

I agree that it looks ugly. Maybe a bit overengineered?

> Do we really need to separate namespace (e.g. "SYM") and attribute (e.g. "NAME")? Maybe we could completely remove the namespace. If still required, the namespace could be integrated into the attribute name instead (e.g. "${SYM_NAME}").

👍 

> For example the attribute "${SYM::NAME}" is a built-in attribute which is implicitly defined by symbols, it's not listed in any attribute list which the user can edit.

Maybe these fields could be added to attribute lists, but greyed out so they're not editable? Or are there too many of them for this to be useful?

ubruhin, 2017-08-20:
> I agree that it looks ugly. Maybe a bit overengineered?

Maybe ;) What about `%NAME`, `%VALUE`, `%PROJECT_AUTHOR` and so on?

> Maybe these fields could be added to attribute lists, but greyed out so they're not editable? Or are there too many of them for this to be useful?

Sounds interesting, but I think there are several problems with that:
- There can be many of such attributes (for example a symbol inherits all attributes from its component, its schematic and its project).
- It may be confusing because some attributes are then shown/editable in different ways (e.g. the name of a symbol is already editable with a separate text input field because it is probably the most important attribute of a symbol, much more important than any other attribute).
- The implementation would require some additional effort (GUI programming is a pain...).

I think the simplest solution would be to just ignore conflicts. But we would need to define which attribute would have the higher priority in case of a conflict: the built-in ones or the user defined ones?

dbrgn, 2017-08-20:
Conflicts could be indicated in the UI (e.g. showing an exclamation mark, with a mouseover text saying that this field is conflicting another field).

ubruhin, 2017-09-08:
I have started to document how the attributes system could look like: [Attributes System](http://librepcb.org/LibrePCB-Doxygen/refactor_attributes_system/de/df4/doc_attributes_system.html)

(If the direct link is broken, click on "Attributes System" [on this page](http://librepcb.org/LibrePCB-Doxygen/refactor_attributes_system/))

ubruhin, 2017-09-10:
Now the new [attributes system specifications](http://librepcb.org/LibrePCB-Doxygen/refactor_attributes_system/de/df4/doc_attributes_system.html) should be more or less complete.

The new system requires to change following things of the current attributes system:
- Remove support for namespaces
- Change syntax from `${NAMESPACE::ATTRIBUTE}` to `#ATTRIBUTE`
- Add attributes to Device library elements and devices in boards
- Change [lookup order](http://librepcb.org/LibrePCB-Doxygen/refactor_attributes_system/de/df4/doc_attributes_system.html#doc_attributes_system_lookup_order)
- Add support for [multiple key substitution](http://librepcb.org/LibrePCB-Doxygen/refactor_attributes_system/de/df4/doc_attributes_system.html#doc_attributes_system_multiple_key_substitution)

@dbrgn @rnestler What do you think about the new attributes system?

rnestler, 2017-09-11:
The description for the "#VALUE" attribute was confusing for me. In the first example it sounds like it is only used for numeric values like 100nF or 10Ohm. This *wrong* perception is encouraged in the diagram, which again shows 100nF for "#VALUE"
When I reached the point of "Multiple Key Substitution" I was very confused because it suddenly introduced the concept of using "#VALUE" for "the most important attributes".

rnestler, 2017-09-11:
I had an idea for placement annotations: Sometimes one adds a comment on the schematic like: "Place close to VDD pin" on a capacitor.
It would be nice if this comment was visible when lay outing. For example as a hint when hovering or when grabbing the part.

ubruhin, 2017-09-11:
> The description for the "#VALUE" attribute was confusing for me.

Should be less confusing now :)

ubruhin, 2017-09-22:
I have now added the documentation about [manual end of a key](http://librepcb.org/LibrePCB-Doxygen/refactor_attributes_system/de/df4/doc_attributes_system.html#doc_attributes_system_manual_end).

And most of the documented features are now already implemented, works pretty good!

But there are still some open questions. For example I'm not sure how schematic frames should work exactly. Currently, schematic frames consist of a symbol and a component like every other component. But the main difference is that they don't show the attributes `#NAME` and `#VALUE`, but other things. Here an example:
[image]

When this component/symbol is added to a schematic, it looks like this:
[image]

This works as long as the user doesn't need the ability to override these built-in attributes. But I think sometimes there are cases which require to override some of these attributes. Just one stupid example: The project name is too long to be displayed in the schematic frame. Then the user may want to add one more attribute, like `#PROJECT_SHORT` which contains the project name in a shorter form. This attribute should then be shown in the schematic frame (instead of `#PROJECT`).

One possible solution would be to use the [multiple key substitution feature](http://librepcb.org/LibrePCB-Doxygen/refactor_attributes_system/de/df4/doc_attributes_system.html#doc_attributes_system_multiple_key_substitution). This could then look like this:
[image]
So as long as there is nowhere an attribute `#FRAME_PROJECT` defined, the fallback `#PROEJCT` is used. If the user wants to override the project name shown in the frame, he just needs to define the attribute `#FRAME_PROJECT` somewhere (at component, schematic or project scope). This way, the labels in the symbol can be decoupled from the built-in attributes.

But as you can see, this makes the labels unnecessary long and thus the symbol looks ugly ;)

Another way would be to use shorter labels which only differ slightly from the built-in attributes, e.g:
[image]

All the used attributes (with underscore at the begin) need then be defined somewhere. This could be done in the library component, with user defined attributes:
[image]

This way, we probably have the highest flexibility as all the displayed attributes can be overridden even at component level, so the attributes in the symbol are completely decoupled from the built-in attributes. And they are still very easy to use because the user does not need to define attributes by himself, as long as he's happy with the default attributes. The only drawback may be the higher complexity which may be confusing...

ubruhin, 2017-09-24:
...another question is if the attributes from `library::Component` should be *copied* to `project::ComponentInstance` when adding a new component to the schematic, of if the `project::ComponentInstance` should only *inherit* the attributes from `library::Component`. Same question applies to `library::Device` and `project::BI_Device`.

Especially when replacing components/devices by another library element, it makes a difference how attributes are handled depending on if they are copied or inherited.

But I think we can continue with the current solution (copying the attributes) and see how this works in practice. If it doesn't work good, we can change this behavior some time...

PR #184 (2017-09-30): Refactor attributes system
Fixes #171.
closed 2017-09-30 by ubruhin