LibrePCB/LibrePCB #175: Where do the continuous builds get uploaded to?

probonopd, 2017-08-13:
Where do the continuous builds get uploaded to? I only found https://bintray.com/librepcb/LibrePCB-Nightly/download_file?file_path=LibrePCB-Nightly-Linux-x86_64.AppImage but this seems to be rather old.

ubruhin, 2017-08-13:
Oh that's strange, it seems that Travis has stopped to build the AppImage and thus it couldn't be uploaded to BinTray :(

But I have no idea why the AppImage is no longer built...

probonopd, 2017-08-13:
The build log says:

> Installing deploy dependencies
> [Bintray Upload] Reading descriptor file: ./dev/travis/bintray.json
> [Bintray Upload] Warning: Path: LibrePCB-x86_64.AppImage does not exist.
> [Bintray Upload] Publishing version 'nightly' of package 'LibrePCB-Nightly-Linux-x86_64'...
> [Bintray Upload] Bintray response: 403 Forbidden. Files from version 'nightly' can't be published. Files can only be published within 180 days from the version publish date.

I think you need to delete the version "nightly" on Bintray and create it again. Then re-run the build.

dbrgn, 2017-08-13:
Maybe a new release could be created through an API?

ubruhin, 2017-08-13:
The main problem is that the AppImage was not even build by Travis:

> [Bintray Upload] Warning: Path: LibrePCB-x86_64.AppImage does not exist.

I found that the link [https://github.com/probonopd/linuxdeployqt/releases/download/3/linuxdeployqt-3-x86_64.AppImage](https://github.com/probonopd/linuxdeployqt/releases/download/3/linuxdeployqt-3-x86_64.AppImage) is no longer valid, so Travis cannot download `linuxdeployqt` and thus cannot build the AppImage. But actually I wonder why the build still succeeds even if something goes really wrong?!

And @probonopd did you delete the `linuxdeployqt` AppImage? Or do you host it somewhere else?

probonopd, 2017-08-13:
Please use https://github.com/probonopd/linuxdeployqt/releases/download/continuous/linuxdeployqt-continuous-x86_64.AppImage

ubruhin, 2017-08-13:
Why are the older versions no longer available? I don't like to always use the latest version because it could break my CI at any time and thus forces me to immediately fix such problems. I prefer to use a fixed version instead which I manually update from time to time.

probonopd, 2017-08-13:
The old versions had too many issues. I could do another fixed version if you really require it, but I am using `continuous` for all my AppImages, so it is highly likely that it is working. ;-)

ubruhin, 2017-08-14:
Ok I'll update to the continuous build for now and hope that it works as expected ;)

probonopd, 2017-08-14:
I think you still need to delete the version "nightly" on Bintray and create it again. Then re-run the build.

closed 2017-08-14 by ubruhin

ubruhin, 2017-08-14:
Unfortunately it's not even possible to delete a package from BinTray after 180 days, so I had to delete the whole "repository" (this is how it's called on BinTray). That worked (i.e. the AppImage on BinTray is up-to-date again), but it's really a pain to do that every 180 days ;)

I opened #179 to work on this new problem.

PR #178 (2017-08-14): Travis: Update to latest linuxdeployqt
Fixes #175.