LibrePCB/LibrePCB #179: BinTray: Files can only be published within 180 days from the version publish date.

ubruhin, 2017-08-14:
Travis uploads a new AppImage after every build on `master`. But 180 days after the first upload, the file cannot be overwritten or even removed again:

```
[Bintray Upload] Publishing version 'nightly' of package 'LibrePCB-Nightly-Linux-x86_64'...
[Bintray Upload] Bintray response: 403 Forbidden. Files from version 'nightly' can't be published. Files can only be published within 180 days from the version publish date.
```

This is unusable for nightly builds!

So: Is there any workaround for this? Or is there another hosting service which is better suited? Actually I don't like BinTray anyway because it is far too "complicated" just for hosting some files. A much more simple hosting service may be better suited for our use-case.

probonopd, 2017-08-15:
Bintray is meant for released versions, not for continuous or nightly builds (they have a paid version for that). As a possible alternative, you might check out uploading to GitHub Releases. I have a little [tool](https://github.com/probonopd/uploadtool) for it.

dbrgn, 2017-08-15:
You could also just upload the files to your own cloudscale server :) As long as LibrePCB is not (yet) incredibly popular, that will be just fine.

ubruhin, 2017-08-15:
> As a possible alternative, you might check out uploading to GitHub Releases.

The disadvantage of this solution is that all watchers of the repository will receive an email every time a release gets updated. For me that's too much spam ;) Or do you know a way to avoid the spam?

> You could also just upload the files to your own cloudscale server :)

Would be possible, but I think as soon as the server is really used as API server, that's probably not the best solution because of security and performance reasons ;)

I think @probonopd needs to introduce an AppImage hosting platform :stuck_out_tongue_winking_eye:

probonopd, 2017-08-15:
I think watchers actually welcome this kind of "spam" (otherwise they wouldn't be watching)?

> I think @probonopd needs to introduce an AppImage hosting platform

Hey, I am not Canonical ;-)

ubruhin, 2017-08-15:
> I think watchers actually welcome this kind of "spam" (otherwise they wouldn't be watching)?

For "normal" releases (which are created/updated very rarely), I agree because that's an important information. But automatic builds for every commit on `master` are not really important and occur too often, so IMO that's not worth to send emails for that ;)

dbrgn, 2017-08-15:
> but I think **as soon** as the server is really used as API server

...that might take a while. Once that happens, we can think about the hosting again.

ubruhin, 2018-05-22:
We no longer need BinTray because of #265. I will remove our files on BinTray soon.

closed 2018-05-22 by ubruhin