LibrePCB/LibrePCB #181: Improve reusability of graphics items

ubruhin, 2017-08-31:
The code for drawing some graphics items like footprints or symbols exist multiple times because of slightly different requirements.

For example a resistor shown in the library editor looks like this:
[image]

And when adding the same symbol (as part of a component) to a schematic, it looks different:
[image]

To avoid code duplication it would make sense to make these graphics items more generic, so they can be used in different situations.

dbrgn, 2017-09-01:
Good idea. But observe the [rule of three](https://erikbern.com/2017/08/29/the-software-engineering-rule-of-3.html) :slightly_smiling_face:

ubruhin, 2017-09-23:
Just found a bug which could be solved when refactoring the graphics items: The rendered text height in the library editor is not exactly the same as in the project editor.

> But observe the rule of three

Should be observed :smile: For example the code to render a symbol exists three times:
- [librepcb::library::SymbolGraphicsItem](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcb/library/sym/symbolgraphicsitem.cpp)
- [librepcb::library::SymbolPreviewGraphicsItem](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcb/library/sym/symbolpreviewgraphicsitem.cpp)
- [librepcb::project::SGI_Symbol](https://github.com/LibrePCB/LibrePCB/blob/master/libs/librepcb/project/schematics/graphicsitems/sgi_symbol.cpp)

dbrgn, 2017-09-23:
:ok_hand:

PR #969 (2022-04-18): Fix duplicate symbol graphics items code & improve symbol renderings
- Refactor `SymbolGraphicsitem` and `SymbolPinGraphicsItem` to support rendering of component previews and to move them from the `core` library to the `editor` library (see #67).
- Remove `SymbolPreviewGraphicsItem` and `SymbolPinPreviewGraphicsItem` since they are now replaced by `SymbolGraphicsitem` and `SymbolPinGraphicsItem`.
- Refactor `SGI_Symbol` and `SGI_SymbolPin` to avoid duplicated code for rendering - now it uses other, existing classes to do the rendering.
- Remove all debugging graphics layers (I didn't use them [... cut]
closed 2022-04-18 by ubruhin