LibrePCB/LibrePCB #188: Define development process used after considering file format as stable

ubruhin, 2017-10-06:
Currently the development is done on the `master` branch, where breaking changes of the file format can occur at any time. An upgrade mechanism (see #185) may only be provided to upgrade one stable file format version to the next stable file format version, but not to upgrade intermediate (unstable) file format versions. So libraries and projects which were created with such a development snapshot may get completely useless (can no longer be opened) in future. This is *very* dangerous for end users!

I see two possible ways how to proceed developing after considering the file format as stable:
- Continue developing on the `master` branch, but warn users to *never* productively use LibrePCB compiled from the `master` branch. Not sure if this works (who reads such warnings...) :wink: 
- Develop on another branch (e.g. on `develop` as suggested in [git flow](http://nvie.com/posts/a-successful-git-branching-model/)) and use `master` only for stable releases. So LibrePCB compiled from the `master` branch is not dangerous and can be used productively at any time.

rnestler, 2017-12-09:
Wouldn't it be easiest to just provide upgrade paths from the released version to the current master?

So one could work on ones project with a released version and for testing open it with the current master version to check if the import works and to test the latest features.

ubruhin, 2017-12-09:
> Wouldn't it be easiest to just provide upgrade paths from the released version to the current master?

Of course, we always need to provide an upgrade path from the last stable release to the current development snapshot (i.e. `master` or `develop`). But we will *not* provide an upgrade path from the current development snapshot to the next stable release, so this upgrade path basically ends in a dead-end ;) And *this* is what we need to avoid to (accidentally) happen for end users.

rnestler, 2017-12-12:
@ubruhin I would consider master a development snapshot and only tags to be (semi-)stable releases. I also think git flow is a bit over engineering and not really useful if you don't plan to maintain several stable versions.

dbrgn, 2017-12-12:
I agree with @rnestler, but it would be good to show a warning in non-release builds.

dbrgn, 2017-12-12:
By the way, even though `master` gives no guarantees, I'd still expect file format experimentation / development to happen in feature branches, so it's probably rare that a file format change is done and then reverted or immediately changed again in master.

ubruhin, 2017-12-12:
Yep I agree too, I think we should continue developing on feature branches which are then merged into `master`. So the file format on `master` is always considered as unstable. But because of that, we really should add a scary warning, maybe in two ways:

- A comment in README.md
- Compiled into the application, for example a big scary warning in the Control Panel.

rnestler, 2017-12-13:
@ubruhin I would suggest to show a big scary warning when one actually tries to convert a project with a master build and to recommend there to save it at a new place and keep the old version.

ubruhin, 2018-09-07:
> I would consider master a development snapshot and only tags to be (semi-)stable releases. I also think git flow is a bit over engineering and not really useful if you don't plan to maintain several stable versions.

@rnestler I think it's not that easy because of the file format versions. If I create the tag v1.0 on the master branch, and afterwards a new feature which changes the file format is merged into master, I would no longer be able to create a v1.1 release on master. The v1.1 release *must* be based on the v1.0 tag, so in that case we would need a release branch.

I agree that generally we should not provide updates for already released major versions, but sometimes it may still be necessary (e.g. critical bugfixes are needed, or many new features are ready which don't need a file format upgrade).

Here an idea for a development workflow:

[image]

In some situations it can be simplified, for example the branches `master` and `release/0.1` can be identical (fast-forward merged) as long as there are no file format changes on master. But as soon as there is a file format change on master, the release branch must diverge.

> I would suggest to show a big scary warning when one actually tries to convert a project with a master build and to recommend there to save it at a new place and keep the old version.

That would be nice, but requires much more effort to implement than just a warning...

ubruhin, 2018-09-07:
Another thing to consider is the `git describe` string which is compiled into the binary. Actually I don't like it because of several reasons:
- It breaks reproducible builds. If you build the exactly same commit again, the output may be different just because some tags were added or removed.
- If tags are created on release branches, its output on the master branch is meaningless.
- If there are tags on both master and release branches, its output on the master branch may even be wrong (outputs the last tag which was created on master, even if there are newer tags on release branches). Currently it outputs the tag "first_pcb" which is already annoying...

Maybe it would be better to replace the `git describe` string with the commit hash because that's the only reliable data which can be extracted from Git. Even branch name and tag name are not reliable because on CI the HEAD is typically set to a commit hash instead of a branch or tag.

dbrgn, 2018-09-07:
> It breaks reproducible builds.

Is this a goal? Are builds reproducible without the git describe link?

> Maybe it would be better to replace the git describe string with the commit hash

Yes, why not :) If you just want a descriptive string, then `git describe` may be ok, if you want a stable reference then the git commit hash would be better.