LibrePCB/LibrePCB #196: Add support for copper planes in boards

ubruhin, 2017-11-22:
Copper planes are actually quite important, but not yet implemented in LibrePCB. Probably the most complicated part of planes is to calculate all segments of the plane, while every segment must satisfy various requirements:

- Keep clearance to other nets or objects (according to design rules)
- Ensure a minimum width at any point in the plane (remove plane segments which are too thin)
- Optionally connect pads by thermal relief
- Optionally remove segments which are not connected to any net (orphans)

I think this can basically be done with following steps:
1. Take plane outline polygon:
[image]
2. Take the outline of every object which must not be connected to the plane, and grow these outlines by their required clearance width:
[image]
3. Subtract these objects from the outline:
[image]
4. Expand (offsetting) the resulting segments by the minimum width value:
[image]
5. Shrink (offsetting) the resulting segments by the minimum width value:
[image]
6. Fill the resulting segments:
[image]
7. Result (No thermal pads, keep orphans):
[image]

I used the [clipper library](http://angusj.com/delphi/clipper.php) for these operations, seems to work pretty good. The only disadvantage is that it doesn't support arcs in polygons, only straight lines are allowed. So we need to convert arcs to many small segments of straight lines :( Unfortunately I didn't find a library which can handle arcs...

Btw, I found some interesting papers about polygon offsetting:
- https://seant23.files.wordpress.com/2010/11/anoffsetalgorithm.pdf
- http://www.me.berkeley.edu/~mcmains/pubs/DAC05OffsetPolygon.pdf

dbrgn, 2017-11-22:
Awesome! 🎆

sebgonzo, 2017-11-23:
Great work!
Don't worry not supporting arcs right now. I'm sure that Altium is also using lines instead.
I think that you should think about unconnected copper islands removal function or at least option to manually place polygon cutouts.

dbrgn, 2017-11-23:
Yeah, cutouts / keepout areas are important, but probably pretty simple to implement once we have the copper plane feature.

nimish, 2017-11-24:
CGAL can handle circular arcs, and there are fast and accurate algorithms for approximating bezier curves with circular arcs and lines, if you need that.

nimish, 2017-11-27:
Hmm, I took a look through this problem. CGAL discussed adding support for polygons with circular arcs here http://cgal-discuss.949826.n4.nabble.com/Inset-and-offset-of-General-polygon-set-2-with-circle-arcs-td4661729.html for the offset operation. Unfortunately, it seems to be dead.

ubruhin, 2017-11-27:
Yes, it seems to be really hard to find a library which can perform polygon offsetting with support for circular arcs :( For the moment I will just use the Clipper library, we could still switch to another library some time if there is one which supports circular arcs...

> I think that you should think about unconnected copper islands removal function

I'm working on that :)

ubruhin, 2017-11-30:
> I suspect you could use an approximation algorithm like Tiller and Hanson?

No, LibrePCB doesn't use beziers but only circular arcs which are pretty easy to approximate by linear line segments ;)

kliment, 2018-02-07:
Looking through the kicad and horizon codebase they also segment arcs.

PR #239 (2018-02-21): Add initial support for copper planes in boards
This adds an initial implementation for copper planes:

[image]

There are still some limitations:
- No clearance values from design rules (because we don't yet have design rules for that)
- No thermal pads (because it seems to be quite complicated to calculate them, so I'm not yet sure how to implement it exactly)

But I think even with these limitations, planes are quite usable for many PCB projects, so I think we can fix them some time later ;)

Fixes #196
closed 2018-02-21 by ubruhin