LibrePCB/LibrePCB #197: how to add component and there's no help doc or wiki for instructions

243750496, 2017-11-25:
[image]

1.how to add component
i have done installed all library 
[image]

2.there's no help doc or wiki for instructions
where can i get the help doc or articles about how to use librepcb ?

3.how to convert kicad library to librepcb?
https://github.com/tmk/kicad_lib_tmk
i need this library for me to working on creating my own keyboard .

thx a lot

ubruhin, 2017-11-25:
Hi,

These libraries are currently all empty because the file format is not yet considered as stable. You can get working libraries by using the [demo workspace](https://github.com/LibrePCB/demo-workspace). See [the Workspace section in the bottom of README.md](https://github.com/LibrePCB/LibrePCB#workspace):

> Note: As LibrePCB is still under heavy developement, it's highly recommended to use this demo workspace. Otherwise you won't be able to create/use any components and some parts of the application will not work properly.

I will think about how to improve this, because it's kind of annoying for new users.

> 2.there's no help doc or wiki for instructions

That's true, currently the only "getting started instructions" are in the README.md, but these needs to be improved a lot.

> 3.how to convert kicad library to librepcb?

Thats not (yet) possible. Converting Eagle libraries is kind of possible, but the converter is not yet integrated into LibrePCB. But please note that LibrePCB uses a library concept which is quite different from KiCad and Eagle, and thus converted libraries will never be as powerful as native LibrePCB libraries.

243750496, 2017-11-29:
i create my own libraries but can not find anywhere to add components from my lib (not empty)
and 
then i tried to copy it to new working space and load it , but still can't add any components of mine (create by myself , not copy and not reedit ),
at last i try to recreate all and delete demo working space but with try several times of restart still get a empty list 

could you please help me?

ubruhin, 2017-12-01:
Replaced by #199 and #205.

closed 2017-12-01 by ubruhin