LibrePCB/LibrePCB #198: can not delete any board!!!!!!!!!!!!!!!!!!!!!!!!!!!!

243750496, 2017-11-26:
after add a board by accident and try to delete it i found nowhere to delete the board 
if i deleted the file , i will never open up the project 
here's the image
[image]
[image]

ubruhin, 2017-11-26:
This is not yet implemented. When deleting the board file manually, you also need to remove the board from the file core/boards.lp. Then you should be able to open the project again.

ubruhin, 2017-11-26:
Fixed in 7fc2cb861fabc5965c5cb31c0dcda484b0fd2ac0 -> removing boards is now possible.

closed 2017-11-26 by ubruhin