LibrePCB/LibrePCB #199: Remove need for demo workspace

ubruhin, 2017-11-26:
I was already asked several times why the [libraries from the library manager](https://github.com/LibrePCB-Libraries) are completely empty. It's quite annoying that the library manager basically works, but is still completely useless until now.

I think we should pick the "higher quality" library elements from the demo workspace and add them to these libraries. For users this should already be enough to remove the need for the demo workspace. 

Of course the file format is still not stable, so if a user fetches the latest libraries with an older version of LibrePCB, the libraries may no longer work. But at least there is a possibility that the versions match, and thus the libraries work properly :wink: 

Steps to do:
- [ ] Add library elements to our libraries: https://github.com/LibrePCB-Libraries
- [ ] Update our *very dynamic* API: https://github.com/LibrePCB/librepcb-api-static/blob/master/api/v1/libraries
- [ ] Update instructions in `README.md`

Btw, are we allowed to publish libraries with the name of a company (e.g. [Molex.lplib](https://github.com/LibrePCB-Libraries/Molex.lplib))? Should we at least add the word "unofficial" to the description (even if it is an "official LibrePCB library")?

ubruhin, 2017-12-22:
Done -> LibrePCB can now be used without the demo workspace.

closed 2017-12-22 by ubruhin

dbrgn, 2017-12-22:
Great :)