LibrePCB/LibrePCB #200: can not remove anything , it's a bug!

243750496, 2017-11-29:
[image]
hope you can fix

ubruhin, 2017-11-29:
Oh that's a misleading GUI. These menu items should be disabled if no library element is open. To remove whole library elements, currently the only way is to open it and then choose File -> Remove Element.

243750496, 2017-11-29:
find still no able to delete anythins with File -> Remove Element. and update library
 **maybe it's a bug**
[image]

243750496, 2017-11-30:
[image]
this is the rightway to do delete , in main page after opened the library manager , we can not delete it 

only can delete it with opened the item , please correct it ,  it's a worg guide to show up in main page without function

ubruhin, 2017-11-30:
@0xB767B Maybe you could improve the GUI to avoid confusion? Most important would be to disable/enable some menu items depending on the current state of the library editor.

0xB767B, 2017-12-01:
@ubruhin Are there any general concepts in LibrePCB how to enable/disable menu-items based on some state of the program?

ubruhin, 2017-12-01:
Not really. I guess the library editor is currently the only window which enables/disables some `QAction`s depending on which kind of library element is currently open. And the implementation of that is quite pragmatic ;)

Btw, now I noticed that some `QAction`s in the toolbars are disabled properly, but not their corresponding menu items (most `QAction` objects are added to both, a toolbar and a menu).

lukabosnjakovic, 2018-06-08:
I had similar problem few minutes a go. It is not intuitive that you must first open element and than go to Edit > Remove. My proposal is that maybe you add context menu (right click on element) with remove option. This is more intuitive to me.

dbrgn, 2018-11-16:
I'd nominate this for the 0.1 final :slightly_smiling_face:

ubruhin, 2018-11-16:
I'm ready to accept pull requests ;)

ubruhin, 2018-12-02:
Context menu added in #380.

> Btw, now I noticed that some QActions in the toolbars are disabled properly, but not their corresponding menu items (most QAction objects are added to both, a toolbar and a menu).

Fixed in c0d39eb020d2b6c839a18fb7a0af1f5764d13dfd.

closed 2018-12-02 by ubruhin