LibrePCB/LibrePCB #201: after import library there's no component there

243750496, 2017-11-29:
Steps:
1.i create my own libraries but can not find anywhere to add components from my lib (not empty)
and 

2.then i tried to copy it to new working space and load it , but still can't add any components of mine (create by myself , not copy and not reedit ),

3.at last i try to recreate all and delete demo working space but with try several times of restart still get a empty list 

Q:could you please help me?

[image]

# btw: i hope you can add import from outside lib folder function to get easy share the 3-party-lib with others

243750496, 2017-11-30:
sloved by create a component at second of creation 
first: create a symbol
then:create a component
final : we can see it in libraries

btw:if more useful please set keywords in it

243750496, 2017-11-30:
Q:what is required? what's the different between check the mark or not check it?
[image]

243750496, 2017-11-30:
get puzzled much more:
[image]

[image]

NQ1: why symbol pin can't edit to change it's name?
NQ2: what's the two list in image means 
NQ3: what can they do for me if i choose one for each of them?just an sign for nothing or done something witch is very important of designing?

243750496, 2017-11-30:
[image]

2More Question 1 :what is Footprint Variants means ?why it can add more ? will it be conflicated with previous one?if not what will happpend if i set two of them?

2:why we need set pad name ?

ubruhin, 2017-11-30:
I'm sorry, I can't explain every detail here and now. But basically you can just use the default values, they are fine for most cases. If you still have questions, please ask us in [our IRC channel](https://webchat.freenode.net/?channels=#librepcb).

Basically we need to create documentation for end users to clarify all of your questions. I created #205 to work on that.

I guess this issue could be closed now (?).

closed 2017-11-30 by ubruhin

243750496, 2017-11-30:
i hope you can add import from outside lib folder function to get easy share the 3-party-lib with others
not just copy the library to the working space 
i mean maybe we can pack it up and use it to import  instead